Nachtzuster/BirdNET-Pi #2: Error messages when changing language in settings

Svardsten53, 2024-02-16:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
When changing database language to Swedish I got two error messages on the front page. Changing back to English and also clear all data doesn't help.

**To Reproduce**
Change language to Swedish in settings.

**Expected behavior**

**Screenshots**
Fatal error: Uncaught TypeError: array_map(): Argument #2 ($array) must be of type array, bool given in /home/landsort/BirdNET-Pi/scripts/overview.php:161 Stack trace: #0 /home/landsort/BirdNET-Pi/scripts/overview.php(161): array_map() #1 {main} thrown in /home/landsort/BirdNET-Pi/scripts/overview.php on line 161

Fatal error: Uncaught TypeError: array_map(): Argument #2 ($array) must be of type array, bool given in /home/landsort/BirdNET-Pi/scripts/overview.php:161 Stack trace: #0 /home/landsort/BirdNET-Pi/scripts/overview.php(161): array_map() #1 {main} thrown in /home/landsort/BirdNET-Pi/scripts/overview.php on line 161

**Additional context**

**Your build**
Raspberry Pi 4B 8GB Bookworm Lite 64-bit

**Code or log snippets**

Nachtzuster, 2024-02-16:
Interesting, it tries to read `scripts/blacklisted_images.txt` on that line
https://github.com/Nachtzuster/BirdNET-Pi/blob/53633440c64d161669ca3f309815bf68afa275d7/scripts/overview.php#L161

could you run these commands and report back?:
`awk -F: '/1000/{print $1}' /etc/passwd`
`awk -F: '/1000/{print $6}' /etc/passwd`
`ls -l scripts/blacklisted_images.txt`
`cat scripts/blacklisted_images.txt`

Svardsten53, 2024-02-16:
I have replaced the real username with "test" in this output.

```
birdnetpitest:~/BirdNET-Pi$ awk -F: '/1000/{print $1}' /etc/passwd
test
birdnetpitest:~/BirdNET-Pi$ awk -F: '/1000/{print $6}' /etc/passwd
/home/test
birdnetpitest:~/BirdNET-Pi$ ls -l scripts/blacklisted_images.txt
ls: cannot access 'scripts/blacklisted_images.txt': No such file or directory
birdnetpitest:~/BirdNET-Pi$ cat scripts/blacklisted_images.txt
cat: scripts/blacklisted_images.txt: No such file or directory
birdnetpitest:~/BirdNET-Pi$ whoami
test

```

Nachtzuster, 2024-02-17:
Thanks - that looks fine.
I can't reproduce though.
So, could you provide the installation log? it sits in the users home directory.

Sparkybreno, 2024-02-18:
I have the same error messages when I enter my api key

Searchyourname, 2024-02-18:
To fix this I just made that  .txt file. It was missing for me.

Sparkybreno, 2024-02-18:
> To fix this I just made that .txt file. It was missing for me.

Genius, that worked

Searchyourname, 2024-02-18:
> I did a new installation from scratch. The error messages are not related to the language but to the **Flickr API key**, which I copied from my other installation. When I removed the key the error messages disappeared.

To fix this I just made the blacklisted_images.txt file. It was missing for me.

Svardsten53, 2024-02-18:
Perfect, problem solved by adding that file. 
[Nachtzuster](https://github.com/Nachtzuster): Please add solution to the installation.
I close this issue.

closed 2024-02-18 by Svardsten53

Smo-RBR, 2024-02-25:
Installed Nachtzuster BirdnetPi last night, via curl -s https://raw.githubusercontent.com/Nachtzuster/BirdNET-Pi/main/newinstaller.sh | bash and the error still persists