Nachtzuster/BirdNET-Pi #5: 'Special' characters like {} don't work correctly when sending a test notification

tastyfishes, 2024-02-17:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
When using the "Send Test Notification" button MQTT notifications with characters such as { show as \{

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools -> Settings
2. Add MQTT configuration in the Apprise Notifications Configuration
3. Add the following text as the notification body: `{"Common_Name": "$comname", "Scientific_Name": "$sciname"}`
4. Click "Send Test Notification"
5. Check received notification 

**Expected behavior**
Expect to receive a notification as `{"Common_Name": "$comname", "Scientific_Name": "$sciname"}` instead it comes through as `\{"Common_Name": "New Holland Honeyeater", "Scientific_Name": "Phylidonyris novaehollandiae"\}`

**Your build**
Raspberry Pi 4 Model B 4GB

Nachtzuster, 2024-02-17:
Confirmed.
This is pretty deep in PHP territory it seems, so not sure I'm brave enough to attempt a fix :-)

Nachtzuster, 2024-02-18:
It should work now.
Please update

tastyfishes, 2024-02-18:
Confirmed fixed, thanks!

closed 2024-02-18 by tastyfishes