Nachtzuster/BirdNET-Pi #6: Top 10 stuck at 23:59

Sparkybreno, 2024-02-19:
[image]
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
On the overview page the top 10 stops at 11:59.  It does correct itself at some point during the day.

**Screenshots**
Attached 

**Additional context**
My Timezone is in Australia, possibly related to this

**Your build**
Pi 4B

Searchyourname, 2024-02-19:
I also have a top 10 issue that may be related. The plot chart image switches to the next date at some point during the day leading to a broken image. The next day that image is showing since the dates match up again. Also, maybe I am misremembering but doesn't the top 10 usually go back to 0 the next day? Mine has showed what looks to be the same plot every day, or it's just adding to the previous days data.

Sparkybreno, 2024-02-20:
Looks like oripy may have sorted this in his fork?

Nachtzuster, 2024-02-20:
Thanks for reporting
could you run these commands:
```
stat ../BirdSongs/Extracted/Charts/Combo-2024-02-19.png
stat ../BirdSongs/Extracted/Charts/Combo-2024-02-20.png
timedatectl
```
also could you get the log of chart_viewer.service from before midnight to when it managed to correct itself?

`sudo journalctl -u chart_viewer.service --no-pager -S '2024-02-19 23:50:00'`

Sparkybreno, 2024-02-20:
I have this, I won’t be able to extract the log until I get home from work.
[image]

Nachtzuster, 2024-02-21:
@Sparkybreno Thanks!
Looks like the charts do get generated as expected:
first created early in the morning when the first bird is detected, last updated shortly after midnight.
So, no need to get the log anymore

Looking at the PHP code to get the chart (that is quite the can of worms) I _think_ the timezone is not handled correctly there.

Nachtzuster, 2024-02-22:
@Sparkybreno Could you try out this testing branch?
First reset your remote:
```
git remote remove origin
git remote add origin https://github.com/Nachtzuster/BirdNET-Pi.git
```
then switch to the testing branch:
`./scripts/update_birdnet.sh -b fix_timezone`

Sparkybreno, 2024-02-23:
Day 2, all is well.  It briefly displays yesterday’s table first thing in the morning but is quickly replaced by the correct table.   Happy with that

Nachtzuster, 2024-02-24:
Great. Fix is merged with main now.
To get your install back to the main branch, got to Tools->System Controls->Update

closed 2024-02-24 by Nachtzuster