Nachtzuster/BirdNET-Pi #8: Several problems after installation

Smo-RBR, 2024-02-25:
Installed the Nachtzuster version this night. After installation runs through, the Pi rebootet, and in the log, BirdnetPi complains about missing model/labels.txt, so i unzipped labels_l18n and symlinked labels_de.txt to labels.txt, and Birdnet startet detecting.
i hope i did this right?

Then when i want to save the Tools/Settings, birdnet throws an error: 
Fatal error: Uncaught TypeError: fwrite(): Argument #1 ($stream) must be of type resource, bool given in /home/birdnet/BirdNET-Pi/scripts/config.php:228 Stack trace: #0 /home/birdnet/BirdNET-Pi/scripts/config.php(228): fwrite() #1 /home/birdnet/BirdNET-Pi/homepage/views.php(181): include('...') #2 {main} thrown in /home/birdnet/BirdNET-Pi/scripts/config.php on line 228

Maybe i caused this with my symlinking of the labels.txt?

Smo-RBR, 2024-02-25:
Okay, maybe i found some problem in installation log, while the install scripts says ist has succesfull performed the installation, the log shows, that the install tried to search/install the labels.txt in the wrong home directory:

`+ cd /home/birdnet/BirdNET-Pi/scripts
+ ./install_language_label_nm.sh -l en
unzip:  cannot find or open /home/smo/BirdNET-Pi/model/labels_nm.zip, /home/smo/BirdNET-Pi/model/labels_nm.zip.zip or /home/smo/BirdNET-Pi/model/labels_nm.zip.ZIP.
unzip:  cannot find or open /home/smo/BirdNET-Pi/model/labels_nm.zip, /home/smo/BirdNET-Pi/model/labels_nm.zip.zip or /home/smo/BirdNET-Pi/model/labels_nm.zip.ZIP.
+ exit 0`

It first has the right directory /home/birdnet but the it uses /home/smo for searching the labels zip

`birdnet:~/BirdNET-Pi/scripts$ ./install_language_label_nm.sh -l de
unzip:  cannot find or open /home/smo/BirdNET-Pi/model/labels_nm.zip, /home/smo/BirdNET-Pi/model/labels_nm.zip.zip or /home/smo/BirdNET-Pi/model/labels_nm.zip.ZIP.
unzip:  cannot find or open /home/smo/BirdNET-Pi/model/labels_nm.zip, /home/smo/BirdNET-Pi/model/labels_nm.zip.zip or /home/smo/BirdNET-Pi/model/labels_nm.zip.ZIP.
birdnet:~/BirdNET-Pi/scripts$ whoami
birdnet
birdnet:~/BirdNET-Pi/scripts$ pwd
/home/birdnet/BirdNET-Pi/scripts
birdnet:~/BirdNET-Pi/scripts$`

Smo-RBR, 2024-02-25:
Sorry, edit the command above, because is had the same idea :)

Nachtzuster, 2024-02-25:
Yes, your analysis is correct.
BirdNET-Pi  does require to be installed as the primary user (uid=1000).
I'm not sure why looking in /etc/passwd was chosen to get to the user and home dir

Anyway, you can set the language though the settings page:
[image]

Smo-RBR, 2024-02-25:
This seems to fail too, because ist uses the same scripts i think. I will use another, fresh SD card for this time testing. Do you plan a fix for this issue?

Nachtzuster, 2024-02-25:
Yes, best try with a fresh cd card.
Getting rid of nosing in /etc/passwd is indeed on my to-do list.
For now I've merged some checks I already had in a branch. I had not merged them a I was not sure it was really needed :-)

closed 2024-02-25 by Nachtzuster