Nachtzuster/BirdNET-Pi #9: Not an issue, so much as praise and a question!

lloydbayley, 2024-02-26:
Firstly, well-done on breathing new life into this project. 

I was very active on the old BirdNet-Pi repo before the 'unpleasantness' and have kept my installation static with my own little changes here and there that are of little or no consequence to others but worked for me. Things like cobbling together an x86 version so I can run it on a Proxmox VM etc etc.

I've have a good look at the additional commits and it's all good stuff and makes a lot of sense. I've been running a parallel system of your repo for a few days now and am noting a few differences here and there (again, nothing to do with your implementation) and it's been working quite well without unexpected calamities! :)

Are you considering opening up the Discussions section for other things so we don't get caught up in a lot of issues that aren't?

All the best,

Lloyd. :)

Nachtzuster, 2024-03-01:
@lloydbayley  Thanks for the kind words.
I'm not entirely sure how I want to go forward with this.
I would prefer all this stuff being merged back, as I don't have the time and energy to manage BirdNET as it used to be.

Anyway I'll open up the Discussions, I guess that won't hurt :-)

Best Regards

lloydbayley, 2024-03-01:
Oh dear. Sorry, I preemptively thought that this might be the new home of this project. I've been looking around and some have tried but not kept up with it and I was refreshed to find yours that was working and recently-updated.

I'd say there is little change of a backward merge as the people in control of that repo aren't interested in working on it anymore. It's a shame but these things happen.

I shall keep an eye on this one to see how things go! Anyway, well-done on the fixes you have made!

Cheers,

Lloyd. :)

Smo-RBR, 2024-03-01:
There is a new project which looks promising: https://github.com/tphakala

jaredb7, 2024-03-07:
> Oh dear. Sorry, I preemptively thought that this might be the new home of this project. I've been looking around and some have tried but not kept up with it and I was refreshed to find yours that was working and recently-updated.
> 
> I'd say there is little chance of a backward merge as the people in control of that repo aren't interested in working on it anymore. It's a shame but these things happen.
> 
> I shall keep an eye on this one to see how things go! Anyway, well-done on the fixes you have made!
> 
> Cheers,
> 
> Lloyd. :)

It's a shame about the other repo, I did wonder what happened after it went quiet. I got busy with life and other projects and figured others just had a break for a while on the other repo but nothing much has happened since. 
As you mentioned though, it happens.

Good to see there's some activity happening in the forks though :), Not to sidetrack discussions but there also https://github.com/Aixsponsa/BirdNET-Pi  which is a fork @Nachtzuster and has fixes here and there (e.g gotty forwards compatibility to help make x86 easier). Which while not necessary, would good to have rolled up into this one too.

Anyways good work on what you've done so far @Nachtzuster

lloydbayley, 2024-03-07:
> It's a shame about the other repo, I did wonder what happened after it went quiet. I got busy with life and other projects and figured others just had a break for a while on the other repo but [...]

Fair play! I had a quick look at that repo you mentioned, yet I see nothing unique beyond this repo. It appears to be a fork of this but there is no self-definition other than having to hunt for differences.

Anyway, yes, it's good to see things are up-to-date as far as OS's and the dreaded gotty :)

closed 2024-03-14 by Nachtzuster