Nachtzuster/BirdNET-Pi #12: Database error loop

Smo-RBR, 2024-03-03:
While browsing a little bit wild through the todays detections, BirdNET pi cames into an error loop and spits out repeatedly only the following error message when accessing the website. Was on mobile, so I could not do further research and sshed to the pi to reboot it. Error message is:

Database is busy
Fatal error: Uncaught Error: Call to a member function execute() on bool in /home/smo/BirdNET-Pi/scripts/overview.php:37 Stack trace: #0 /home/smo/BirdNET-Pi/homepage/views.php(411): include() #1 {main} thrown in /home/smo/BirdNET-Pi/scripts/overview.php on line 37

Nachtzuster, 2024-03-03:
Could check if there are errors in 'View Log' in the ui?
Can you see the data in the database? Tools->Database Maintenance->login->detections->Select Data

Also, could you ssh and run `./scripts/extra_info.sh`

Smo-RBR, 2024-03-03:
Hi,

The log viewer does not show any errors after the reboot. Maybe i rebootet to fast, because i had fear the PI becomes unavailable over network again. And i couldnt find any syslog or messages in /var/log. My linux knowledge is a bit outdated it seems :)

Data in the db seems to be okay and complete. 

birdnet:~/BirdNET-Pi$ ./scripts/extra_info.sh
........................................IPs.....................................
LAN IP: 10.0.1.17
Public IP: 91.57.203.164
..................................`vcgencmd stats`..............................
throttled=0x0
Binary: 0
....................................Clock Speeds................................
arm:    frequency(48)=1200126000
core:   frequency(1)=400000000
h264:   frequency(28)=0
isp:    frequency(45)=0
v3d:    frequency(46)=300000000
uart:   frequency(22)=48000000
pwm:    frequency(25)=0
emmc:   frequency(50)=200000000
pixel:  frequency(29)=338000
vec:    frequency(10)=108000000
hdmi:   frequency(9)=163683000
dpi:    frequency(4)=0
........................................Volts...................................
core:   volt=1.2563V
sdram_c:        volt=1.2000V
sdram_i:        volt=1.2000V
sdram_p:        volt=1.2250V
.....................................Caddyfile..................................
http://  {
  root * /home/smo/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888
}
.................................... Crontab....................................

SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin

17 *    * * *   root    cd / && run-parts --report /etc/cron.hourly
25 6    * * *   root    test -x /usr/sbin/anacron || { cd / && run-parts --report /etc/cron.daily; }
47 6    * * 7   root    test -x /usr/sbin/anacron || { cd / && run-parts --report /etc/cron.weekly; }
52 6    1 * *   root    test -x /usr/sbin/anacron || { cd / && run-parts --report /etc/cron.monthly; }
*/5 * * * * smo /usr/local/bin/disk_check.sh >/dev/null 2>&1
*/3 * * * * smo /usr/local/bin/cleanup.sh >/dev/null 2>&1
@reboot smo /usr/local/bin/cleanup.sh >/dev/null 2>&1
0 1 * * 1 smo /usr/local/bin/weekly_report.sh >/dev/null 2>&1

Nachtzuster, 2024-03-03:
All this looks fine.
but you're still seeing the error in the ui?

Smo-RBR, 2024-03-03:
No, no, sorry, after the reboot it's gone.

Nachtzuster, 2024-03-03:
I see.
The code does look suspect though I can't reproduce it.

Can you reproduce it, and if so, how?

Smo-RBR, 2024-03-03:
I tried, but I can't. If I think about, it must be be triggered by a mix of laggy network connection and fast following db query's. Because I looked for the todays detections, and thought the connection has dropped and touched some kind of wild on the various links. So, maybe it's not such a big deal?

closed 2024-03-14 by Nachtzuster