Nachtzuster/BirdNET-Pi #24: Apprise Issue / Notifications Telegram

Nachtzuster, 2024-03-29:
### Discussed in https://github.com/Nachtzuster/BirdNET-Pi/discussions/23

<div type='discussions-op-text'>

<sup>Originally posted by **MueJosh** March 28, 2024</sup>
Hey,

Yesterday I "Updated" my Birdnet-Pi (the original) to this Fork. (complete new install)

I adapted all my old Settings to this fork.
Now my Telegram Notificantions dont work anymore. If i send a test message it sends the last detections but it doesnt work on its own.
Any Ideas why? 

Any Code needed?

Thanks!
[image]

2. Another Issue but not problematic:
Updating the CONFIDENCE=XX crashes the Tools bar- it updates the new Confidence but i always have to open the "Tools"-Bar again. No Idea why...

</div>

Nachtzuster, 2024-03-29:
@MueJosh This `%` thing is a bug alright, thanks for reporting!

About the second issue: I'm not sure I understand the issue. Could you add some more details, screenshots maybe?
Please do this in another issue.
(Note that I've pushed a bunch of commits earlier today, so there is a slight chance it is fixed by accident)

MueJosh, 2024-03-30:
> Please do this in another issue.
> 
> (Note that I've pushed a bunch of commits earlier today, so there is a slight chance it is fixed by accident)

Fixed by accident ^^ 
or I played too much with my own configs and it works because it's the default again - I'll find out in the next few days.

Thanks for keeping BirdNET-Pi alive!
	(tho it was never dead)

Nachtzuster, 2024-03-30:
I pushed a fix, `%` should work again.

closed 2024-03-30 by Nachtzuster

arne1921KF, 2024-03-31:
@Nachtzuster, would you point me to the lines which caused the problem so I can learn from it?

MueJosh, 2024-03-31:
Though I'm not Nachtzuster:

[https://github.com/Nachtzuster/BirdNET-Pi/commit/cb2969fd3993884c23922d03ce0ffed2a7721000](https://github.com/Nachtzuster/BirdNET-Pi/commit/cb2969fd3993884c23922d03ce0ffed2a7721000)