Nachtzuster/BirdNET-Pi #26: Weekly Report (yes, it's still not working) bug

MueJosh, 2024-04-01:
In the settings menu for Apprise notification there is a problem with the weekly report. 
It doesn't get send (this is also problem in the original repo) 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Tools` - `Settings` - `Notifications`
2. Click on `send weekly report`
3. Error - Don't see a weekly report at the end/ start of the week ;) 

**Expected behavior**
Get the weekly report as an notification from apprise

**Screenshots**
[image]

**Your build**
Raspberry Pi 4
64 bit bullseye lite

**Code or log snippets**

### I think the problem lays in this script: 

> https://github.com/Nachtzuster/BirdNET-Pi/blob/main/scripts/weekly_report.sh

As Its says that the user is named Birdnet-pi and it should be the real username not the default one. 

I can't test it right now but maybe you can get it working.

Nachtzuster, 2024-04-02:
No, that is the directory name in $HOME, and that is ok.

There might be something wrong with your specific setup:
Could you post Tools->System Info->Viewer ?

MueJosh, 2024-04-02:
> No, that is the directory name in $HOME, and that is ok.
> 
> 
> 
> There might be something wrong with your specific setup:
> 
> Could you post Tools->System Info->Viewer ?

Sure, here it is :) 

[image]
[image]
[image]
[image]

Nachtzuster, 2024-04-02:
that looks ok -I think-

Could you run `/usr/local/bin/weekly_report.sh`  manually ? and report the output?
Does that give you a notification?

MueJosh, 2024-04-02:
Yeah, I wasted a lot of time trying to get it running. In the end I gave up (though I would love too see it working) 

[image]

#### And the notification: 
[image]

Nachtzuster, 2024-04-03:
@MueJosh help me understand the issue you're seeing is that 
1) you do not get any weekly notifications or
2) the notification you get are a stack-trace like shown in your screenshot
3) something else

MueJosh, 2024-04-03:
1) Yes, I don't receive a weekly report (though I'm receiving detections etc.) 

2) yep, but I just receive this notification (like in the screenshot) if I send a "test notification" 

I don't even know how the weekly report should look like, is it a picture of the weekly_report.php or plain text? 
I have never seen it before 😅

Nachtzuster, 2024-04-04:
looking at the screenshot, I think you're seeing a division by zero issue.
I pushed a fix to handle that safely.

Could you run `/usr/local/bin/weekly_report.sh` again?
Does that give you a notification now?

MueJosh, 2024-04-04:
Now it gives me nothing 
Normal bird notifications still work 

Updated, reboot, checked the files, checked apprise notifications (saved it again, no changes done)

It always gives me this output (tried sudo once) 
[image]

Really confusing, maybe it's because of the telegram api (idk)

Nachtzuster, 2024-04-04:
This could actually be progress :-)

could you see what `http://[YOURPI]/views.php?view=Weekly%20Report&ascii=true`
gives you, that should be the report

MueJosh, 2024-04-04:
That's nice to hear ^^

[image]

Nachtzuster, 2024-04-04:
Ah yes, looking good. :+1: 
So now the week report itself is working.

I'm not sure about the apprise error though...
Could you remove one of the apprise entries so it should send one single notification and then run
`/usr/local/bin/weekly_report.sh` agiain? _no sudo_

Nachtzuster, 2024-04-04:
hmm, some googling sugests that this could be a Telegram limitation.
similar issue: https://github.com/Eleirbag89/TelegramBotPHP/issues/152

Could you test with another serivice like email oe whatsapp to confirm?

MueJosh, 2024-04-04:
I got fed up with the notifications / weekly report ;) 

Signal needs a docker / server to send messages, email needs the password (I have 2fa enabled- more issues) 
This is why I use telegram 😅 

Problem are all the <>*/ ... on the weekly report 
Adding `?format=markdown` to the chat id "solves" the problem. 
Not great but it works! 

[image]
[image]

[pull request (never did this before; it took some time to figure it out)](https://github.com/Nachtzuster/BirdNET-Pi/pull/30)

Nachtzuster, 2024-04-06:
That is a great find! Seems the best solution I think

arne1921KF, 2024-04-06:
Nice, thank you, @MueJosh.

Interestingly, adding the `?format=markdown` command breaks the formatting of the notification title. 

OTH, it allows formatting for the body, which is great. Having scientific names in _italics_ is such a relief... ;-)
The only thing missing (from markdown, so no solution for that) is the ability to add a new line.

Here's my current take:
`*$comname* (_$sciname_) wurde erkannt. `Confidence score: $confidence, $confidencepct %, cutoff: $cutoff, sensitivity: $sens` [Audio available on local network]($listenurl)`

MueJosh, 2024-04-07:
@Nachtzuster should I open another issue for notification breaks?
I don't mind Handeling it here

Example, this is how it's set up normally:
`Ein/e $comname  ($sciname)  wurde soeben erkannt. Erkennungswahrscheinlichkeit $confidencepct% `
Looks like this: 
[image]

Code for break-line <> , /n , %0A or whatever the server wants for a break
`Ein/e $comname  ($sciname)  wurde soeben erkannt. %0AErkennungswahrscheinlichkeit $confidencepct%`

This is how it's supposed to look like (send with "test notification" button)
[image]

Same break line `%0A`
How it actually comes out: 
[image]

If I have the time (can be weeks to months) I can handle it. 

Maybe it's an issue with apprise? (I don't think so)

Nachtzuster, 2024-04-08:
> should I open another issue for notification breaks?

yes please.

Nachtzuster, 2024-04-09:
@MueJosh I've swapped out the `<small>` tags in the html, it still looks the same as before -kinda- but now sending to Telegram should work without workaround.

closed 2024-04-15 by Nachtzuster