Nachtzuster/BirdNET-Pi #27: Issues with find when using symbolic links.

davidmam, 2024-04-01:
The find command by default does not follow symbolic links. This means that if you want to mount your birdsounds folder on an external drive, the analysis scripts fail. 

The fix is simple, add the -L flag to each find command.

I did put in a pull request on the original repository [Add flags to find to follow symbolic links #942
](https://github.com/mcguirepr89/BirdNET-Pi/pull/942/files) but it seems to have been ignored meaning I have to manually check/fix it every update

Nachtzuster, 2024-04-02:
I'm not sure what it is you are proposing: that file has been factored out in my repo.

If you want the fix applied in https://github.com/mcguirepr89/BirdNET-Pi , then I do not think that will happen since the maintainers have announced they would not continue.

davidmam, 2024-04-02:
The find command is called in a number of places throughout the scripts. 
Would it be possible to add the -L flag to each find command so that symbolically linked mounts will work?

scripts/cleanup.sh:6:empties=($(find ${PROCESSED} -size 57c))
scripts/cleanup.sh:12:if [[ "$(find ${PROCESSED} | wc -l)" -ge 100 ]];then
scripts/cleanup.sh:16:#accumulated_files=$(find $RECS_DIR -path $PROCESSED -prune -o -path $EXTRACTED -prune -o -type f -print | wc -l)
scripts/disk_check.sh:16:        filestodelete=$(($(find ${EXTRACTED}/By_Date/* -type f | wc -l) / $(find ${EXTRACTED}/By_Date/* -maxdepth 0 -type d | wc -l)))
scripts/disk_check.sh:27:        find ~/BirdSongs/ -type d -empty -mtime +90 -delete
scripts/disk_check.sh:28:        find ${EXTRACTED}/By_Date/ -empty -type d -delete;;
scripts/disk_check.sh:30:       #rm -drfv "$(find ${EXTRACTED}/By_Date/* -maxdepth 1 -type d -prune \
scripts/dump_logs.sh:37:CALLS=("df -h" "free -h" "ifconfig" "find ${RECS_DIR}")
scripts/update_birdnet_snippets.sh:19:find $HOME/Bird* -type f ! -perm -g+wr -exec chmod g+wr {} + 2>/dev/null
scripts/update_birdnet_snippets.sh:20:find $HOME/Bird* -not -user $USER -execdir sudo -E chown $USER:$USER {} \+
scripts/update_birdnet_snippets.sh:23:find $HOME/BirdNET-Pi -path "$HOME/BirdNET-Pi/birdnet" -prune -o -type f ! -perm /o=w -exec chmod a+w {} \;

A second issue is the filenames used for the recordings are not acceptable for NTFS file systems (such as a portable hard drive). This again is a simple fix and will easily allow the use of larger data storage without having to reformat the drives. They are then directly removable and readable on Windows systems.

utils/helpers.py:72:        time_created = re.search('[0-9]+:[0-9]+:[0-9]+$', name).group()
utils/helpers.py:73:        self.file_date = datetime.datetime.strptime(f'{date_created}T{time_created}', "%Y-%m-%dT%H:%M:%S")
to
utils/helpers.py:72:        time_created = re.search('[0-9]+_[0-9]+_[0-9]+$', name).group()
utils/helpers.py:73:        self.file_date = datetime.datetime.strptime(f'{date_created}T{time_created}', "%Y-%m-%dT%H_%M_%S")

and 

birdnet_recording.sh: 54-59:
    if [ -z ${REC_CARD} ];then
      arecord -f S16_LE -c${CHANNELS} -r48000 -t wav --max-file-time ${RECORDING_LENGTH}\
	      	      	       --use-strftime ${RECS_DIR}/StreamData/%F-birdnet-%H:%M:%S.wav
    else
      arecord -f S16_LE -c${CHANNELS} -r48000 -t wav --max-file-time ${RECORDING_LENGTH}\
        -D "${REC_CARD}" --use-strftime ${RECS_DIR}/StreamData/%F-birdnet-%H:%M:%S.wav

to
    if [ -z ${REC_CARD} ];then
      arecord -f S16_LE -c${CHANNELS} -r48000 -t wav --max-file-time ${RECORDING_LENGTH}\
	      	      	       --use-strftime ${RECS_DIR}/StreamData/%F-birdnet-%H_%M_%S.wav
    else
      arecord -f S16_LE -c${CHANNELS} -r48000 -t wav --max-file-time ${RECORDING_LENGTH}\
        -D "${REC_CARD}" --use-strftime ${RECS_DIR}/StreamData/%F-birdnet-%H_%M_%S.wav

davidmam, 2024-04-02:
Apologies, markdown has done interesting things to those code snippets.

davidmam, 2024-04-03:
I'd never heard of a bind mount until now. Typically I am linking an external hard drive (/dev/sda1 for example) to ~/BirdSounds. I will try bind mounting and see how that works.

Most external hard drives are preformatted as NTFS which also has the advantage of using them for a yearly archive that is readable across many platforms.  Hence the issue with having friendly filenames (that do not contain : )

Nachtzuster, 2024-04-03:
Not hard to see the advantage of NTFS compatibility. I probably would have gone the the basic IOS8601 format...
But considering the install base of BirdNET-Pi out there, I'm afraid that ship has sailed.

davidmam, 2024-04-03:
The install base is not huge and it is a minor change. Oh well, I will just have to patch manually each time. That makes deployment on a larger scale hard. I'm fairly sure it can be changed without breaking by matching either of two regex instead of just the one.

lloydbayley, 2024-04-06:
With the greatest of respect, the problem I see here is that the original BirdNet-Pi was created as that. Something that is meant to run on a Raspberry Pi. The old repo turned into a pile of beyond the original scope PR's and beyond. The Pi running Linux natively means that supported storage is in a native format.

davidmam, 2024-04-07:
I see your point of view and would respectfully point out that NTFS drives are commonly mounted on most common distros without any additional requirements, and that analysis of data can happen on a variety of systems. It takes a particular frame of mind to demand that the data written on a Pi cannot be easily moved to another system. The whole point of open source is surely to be inclusive, not exclusive.

lloydbayley, 2024-04-08:
ok

Nachtzuster, 2024-04-08:
Enough talk about NTFS, I believe.

@davidmam any update on the original issue? have you been able to test bind mounts in staid of symbolic links?

Also, I'm still not entirely clear on what exactly you are linking: is it just `~/BirdSounds` or more than one link?

davidmam, 2024-04-15:
Though a quick google seems to answer my question.

Nachtzuster, 2024-04-15:
Excellent, thanks for the feedback.

closed 2024-04-15 by Nachtzuster