Nachtzuster/BirdNET-Pi #37: Species stats: "Reviewed" flag settings applied to all observations?

arne1921KF, 2024-04-13:
Could you (and the community) check if this is a bug?

The "Reviewed" flag setting (including the manually set species in case corrected of false positives) seems to carry over to all files, or at least the next sound file (including the next sound file of an unrelated record, i.e. of another species at another recording time).   

To Reproduce:
1. Go to `Species Stats`
2. Click on `Which bird would you like to explore for the dates [...]` dropdown and choose a species. 
3. Choose a recording under `Available recordings`. 
4. Scroll down to `Reviewed`, set combo box to TRUE. 
5. Choose `False Positive` radio button. 
6. Choose the correct species for the chosen audio file.  
7. Scroll up and choose another species, or just another sound file of the same species. Scroll down and see presumed error: `Reviewed` combo box + `False Positive` radio button + previously chosen species are still as set before.

Is this a bug, or a feature?

Expected behavior would be that on change of the sound file as well as change of the species (and thus also the sound file), the combo box, the radio button and the alternatively chosen species are "reset" in the GUI, **but still set for the file I wanted them to applied to. 

Running version [a22c7c089fa94eb56df04d2b5812d28ef379bba4](https://github.com/Nachtzuster/BirdNET-Pi/commit/), on a RPi4B with Bullseye, Kernel Version	6.1.21-v8+.

Just for the record: I ~~highly appreciate~~ love your work, as well as @mcguirepr89's original work. I have my instance running for a couple of weeks now, and I'm planning to go off-grid with a 20 Wp solar panel and a 6Ah LiFePO<sub>4</sub> battery ASAP.  Which could be next Wednesday, actually. \o/

Nachtzuster, 2024-04-15:
The checkbox/radio-buttons seem to be introduced [here](https://github.com/mcguirepr89/BirdNET-Pi/commit/6ef05c2bc3e960045a87f7cefe61737100c5673e) as a kind of mock-up/RFC for a future feature. No followup commits to be found though.

I guess these ui elements should just be removed.

arne1921KF, 2024-04-16:
... or someone who knows a bit about the database (including the ER model to catch potential dependencies...) could try to write a functionality. I assume this should have gone here:

```
#     con.commit()
#     con.close()
```

Ideally, updating the species ID should keep the ID of the originally detected species, because that would add additional value to the data, and a rollback would also be possible.

Disclaimer: 
I'm coming from the R world and have no experience in Python. Also, so far, I haven't looked at the database model at all. If I ever find the time, I would look into that, then learn some basic Python, and then sent you a pull request. But I honestly doubt this would be anytime soon. :-/ 

Suggestion:
You could add a disclaimer to the GUI that this GUI elements don't do anything, and that it would be great if someone could contribute working database updating functionality to the project. What do you think?

Nachtzuster, 2024-04-16:
> Suggestion:
> You could add a disclaimer to the GUI that this GUI elements don't do anything, and that it would be great if someone could contribute working database updating functionality to the project. What do you think?

Since the code has been sitting there for almost two years, I wouldn't hold my breath  :wink: 

I guess no-one saw much value since false positives can just be deleted.
I can't say I do either: it's one thing to write something to the db, doing something useful with it is another. And that is assuming current DB scheme supports it (it doesn't)

Nachtzuster, 2024-04-16:
> I'm planning to go off-grid with a 20 Wp solar panel and a 6Ah LiFePO4 battery ASAP. Which could be next Wednesday, actually. \o/

It'd be great to share it in the show and tell, when you have it up and running. I'm sure more people are thinking about something along these lines.

arne1921KF, 2024-04-16:
> doing something useful with it is another

A low-hanging fruit would be additional stats for total recordings, reviewed recordings and reviewed false positives. 

False positives could also be interesting in context of model training. This isn't in the scope for a broader audience, admittedly. But then, users who are inclined to tinker so much that they run a BirdNet-Pi instead of just downloading the Merlin app or the BirdNET app to their phone are already going into a rabbit hole, aren't they?  ;-)

In regard to reporting back on my off-grid-setup: will do. The panel arrived today, the battery is scheduled for tomorrow. The necessary housing will have to wait for a couple of days.

MueJosh, 2024-04-16:
> In regard to reporting back on my off-grid-setup: will do. The panel arrived today, the battery is scheduled for tomorrow. The necessary housing will have to wait for a couple of days.

OT: 
What kind of bms and/or solar charge controller are you using? 

I've looked at a dozens and they are either way too expensive (for their purpose) or lack basic functions like mppt. There are also some charging hats for the pi but they can't handle 18V (/21V) panels with more than 30W (1,6A) of output... 
for now I'm running battery packs of 30-40 18650 cells and recharge them at home in a few hours.

Nachtzuster, 2024-05-13:
Removed the confusing radiobutton

closed 2024-05-13 by Nachtzuster