Nachtzuster/BirdNET-Pi #41: Apostrophe in Species Name -> Can't Find File

enthusiastpodcast, 2024-04-21:
When a species has an apostrophe in its name, e.g. `Chuck-will's-widow`, the bird is identified correctly. However, when you go to listen to it in `Recordings` -> `By Species`, it returns `No recordings were found.`

The files are saved without an apostrophe in the filename, e.g. as `Chuck-wills-widow-88-2024-04-21-birdnet-00_11_35.wav`, which seems to cause an error when loading Recordings in the web interface.

**Expected behavior**
Recordings should appear in `Today's Detections` and `Recordings` sections of the web interface like other species.

**Your build**
Raspberry Pi 5

lloydbayley, 2024-04-22:
An interesting conundrum. The human in me says to preserve the apostrophe in the name for readability but the logical coder says to abolish them for ease of operation as it can be assumed there may be an apostrophe where it might belong.

Nachtzuster, 2024-04-23:
I pushed a couple of fixes, bird names with quotes should appear again. Not 100% sure I got all the bugs though...

As it turns out, the original BirdNET-Pi was omitting the quotes from bird names when writing to the DB, although _most_ of the PHP code did support the quotes.
Because this fork IS writing bird names as-is @enthusiastpodcast was hitting some ancient latent bugs :-)

lloydbayley, 2024-04-25:
I think the code evolved around other display issues and was never looked at how it writes to the DB...I don't remember any issues to do with that whilst I was involved with the original repo. I've had a look at the code and it's quite the minefield. Well done @Nachtzuster for finding (hopeully all) the bugs.