Nachtzuster/BirdNET-Pi #43: Caddy Page when trying to open the website

dogwood12, 2024-04-25:
I have successfully downloaded the software to my raspberry PI 5 however, when I go to open birdnet in my browser it just takes me to a caddy page with some instructions that I do not understand. I have also tried accessing the page by using my raspberry pi's IP address but this has not worked either and only brings me to the Apache2 page. Not sure where to go from here. I have tried uninstalling and reinstalling a couple of times but nothing seems to be working. 

Thank you

Nachtzuster, 2024-04-26:
Could you upload the installation log?
Since you tried a couple of times, I guess the first one would be of most use.
The log is in your users' home directory

lloydbayley, 2024-04-28:
An installation log would be beneficial. This used to occur with installations on the original repo due to updates that it didn't take into consideration. @Nachtzuster's repo shouldn't have this issue by default. Would be interesting to see what the logs have to offer!

dogwood12, 2024-04-30:
Is this what you are looking for? I don't have much experience with raspberry pi.

+ set -e
+ my_dir=/home/chetanmb/BirdNET-Pi
+ export my_dir=/home/chetanmb/BirdNET-Pi
+ my_dir=/home/chetanmb/BirdNET-Pi
+ cd /home/chetanmb/BirdNET-Pi/scripts
++ uname -m
+ '[' aarch64 '!=' aarch64 ']'
+ ./install_config.sh
+ set -e
+ trap 'exit 1' SIGINT SIGHUP
+ echo 'Beginning ./install_config.sh'
Beginning ./install_config.sh
+ birdnet_conf=/home/chetanmb/BirdNET-Pi/birdnet.conf
+ LATITUDE=0.0000
+ LONGITUDE=0.0000
+ '[' -f /home/chetanmb/BirdNET-Pi/birdnet.conf ']'
+ install_config
+ cat
+ chmod g+w /home/chetanmb/BirdNET-Pi/birdnet.conf
+ '[' -d /etc/birdnet ']'
+ sudo ln -sf /home/chetanmb/BirdNET-Pi/birdnet.conf /etc/birdnet/birdnet.conf
+ grep -ve '^#' -e '^$' /etc/birdnet/birdnet.conf
+ sudo -E HOME=/home/chetanmb USER=chetanmb ./install_services.sh
+ trap 'rm -f ${tmpfile}' EXIT
+ trap 'exit 1' SIGINT SIGHUP
++ mktemp
+ tmpfile=/tmp/tmp.QzdAgq3E5D
+ config_file=/home/chetanmb/BirdNET-Pi/birdnet.conf
+ export USER=chetanmb
+ USER=chetanmb
+ export HOME=/home/chetanmb
+ HOME=/home/chetanmb
+ export PYTHON_VIRTUAL_ENV=/home/chetanmb/BirdNET-Pi/birdnet/bin/python3
+ PYTHON_VIRTUAL_ENV=/home/chetanmb/BirdNET-Pi/birdnet/bin/python3
+ '[' -f /home/chetanmb/BirdNET-Pi/birdnet.conf ']'
+ source /home/chetanmb/BirdNET-Pi/birdnet.conf
++ SITE_NAME=birdnetpi
++ LATITUDE=0.0000
++ LONGITUDE=0.0000
++ MODEL=BirdNET_6K_GLOBAL_MODEL
++ SF_THRESH=0.03
++ BIRDWEATHER_ID=
++ CADDY_PWD=
++ ICE_PWD=birdnetpi
++ BIRDNETPI_URL=
++ RTSP_STREAM=
++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
++ APPRISE_NOTIFICATION_BODY='A  ()  was just detected with a confidence of '
++ APPRISE_NOTIFY_EACH_DETECTION=0
++ APPRISE_NOTIFY_NEW_SPECIES=0
++ APPRISE_WEEKLY_REPORT=1
++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=0
++ APPRISE_MINIMUM_SECONDS_BETWEEN_NOTIFICATIONS_PER_SPECIES=0
++ APPRISE_ONLY_NOTIFY_SPECIES_NAMES=
++ APPRISE_ONLY_NOTIFY_SPECIES_NAMES_2=
++ FLICKR_API_KEY=
++ FLICKR_FILTER_EMAIL=
++ RECS_DIR=/home/chetanmb/BirdSongs
++ REC_CARD=default
++ PROCESSED=/home/chetanmb/BirdSongs/Processed
++ EXTRACTED=/home/chetanmb/BirdSongs/Extracted
++ OVERLAP=0.0
++ CONFIDENCE=0.7
++ SENSITIVITY=1.25
++ FREQSHIFT_TOOL=sox
++ FREQSHIFT_HI=6000
++ FREQSHIFT_LO=3000
++ FREQSHIFT_RECONNECT_DELAY=4000
[... 206 lines cut ...]
+ configure_caddy_php
+ echo 'Configuring PHP for Caddy'
Configuring PHP for Caddy
+ sed -i s/www-data/caddy/g /etc/php/8.2/fpm/pool.d/www.conf
+ systemctl restart 'php7*-fpm.service'
+ echo 'Adding Caddy sudoers rule'
Adding Caddy sudoers rule
+ cat
+ chmod 0440 /etc/sudoers.d/010_caddy-nopasswd
+ config_icecast
+ '[' -f /etc/icecast2/icecast.xml ']'
+ cp /etc/icecast2/icecast.xml /etc/icecast2/icecast.xml.prebirdnetpi
+ sed -i 's/>admin</>birdnet</g' /etc/icecast2/icecast.xml
+ passwords=("source-" "relay-" "admin-" "master-" "")
+ for i in "${passwords[@]}"
+ sed -i 's/<source-password>.*<\/source-password>/<source-password>birdnetpi<\/source-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<relay-password>.*<\/relay-password>/<relay-password>birdnetpi<\/relay-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<admin-password>.*<\/admin-password>/<admin-password>birdnetpi<\/admin-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<master-password>.*<\/master-password>/<master-password>birdnetpi<\/master-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<password>.*<\/password>/<password>birdnetpi<\/password>/g' /etc/icecast2/icecast.xml
+ sed -i 's|

dogwood12, 2024-04-30:
Is this what you are looking for? I am new to raspberry pi. 

[installation-2024-04-30.txt](https://github.com/Nachtzuster/BirdNET-Pi/files/15169615/installation-2024-04-30.txt)

Nachtzuster, 2024-05-02:
Yes, this is the file, thanks!
Looking through the file, nothing really sticks out, except that it is not the one from the first install attempt.

Could you try again on a fresh os image?

dogwood12, 2024-05-02:
After installing a fresh OS and then reinstalling the package, it is up and running. Maybe it had something to do with the pre-installed OS on this particular SD card. Anyways, thank you for the help!

closed 2024-05-02 by dogwood12