Nachtzuster/BirdNET-Pi #44: Gotty not running on new install

tomtom215, 2024-04-27:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**

A new installation on a Raspberry pi 5 8Gb with Bookworm seems to work, but the gotty process shows as not running

**To Reproduce**
Steps to reproduce the behavior:
Fresh install on Raspberry Pi 5 with bookworm

**Expected behavior**
All processes are running as expected

**Your build**
Raspberry pi 5 8Gb with Bookworm

It seems as thought things still work (analysis still runs) but I wanted to open an issue in case there were some troubleshooting options available

tomtom215, 2024-04-27:
`    652 ?        Ss     0:00 bash /usr/local/bin/gotty --address localhost -p 8080 --path log --title-format BirdNET-Pi Log birdnet_log.sh`
`    671 ?        Sl     0:00 gotty-aarch64 --address localhost -p 8080 --path log --title-format BirdNET-Pi Log birdnet_log.sh`
`    679 ?        Ss     0:00 bash /usr/local/bin/gotty --address localhost -w -p 8888 --path terminal --title-format BirdNET-Pi Terminal login`
`    681 ?        Sl     0:00 gotty-aarch64 --address localhost -w -p 8888 --path terminal --title-format BirdNET-Pi Terminal login 1856 pts/0    S+     0:00 grep gotty`

Nachtzuster, 2024-04-27:
Seems to be running fine... odd

Do 'View Log' and 'Tools -> Web Terminal' work?

tomtom215, 2024-04-27:
Oddly it does, and I just successfully tested it, but for some reason Tools -> System Info -> Processes is showing it as not running.

Nachtzuster, 2024-04-27:
Yeah, it is a 'visualization' bug

There is technically no `gotty` process running, but there is a `gotty-aarch64` one ;-)

tomtom215, 2024-04-27:
Cool, thanks for your responsiveness! 

I personally don't rely on the service and don't have an issue as long as everything works, but as this fork seems active, I wanted to report the only "bug" I came across with a fresh install on a RPI5 with Bookworm so I wanted to open an issue to play it safe.

As it is only a "visual" issue, I'll go ahead and close this. 

Thanks again for your help and all f your work on this fork!!!

alexbelgium, 2024-04-28:
Hi, I'm using a script `sed -i "s/,gotty,/,$(ps aux | grep -o 'gotty-[a-z0-9]*' | head -n 1),/g" "$HOME"/BirdNET-Pi/templates/phpsysinfo.ini` to correct the gotty service name

lloydbayley, 2024-04-28:
Now this deserves some further thought. It's a funny old thing isn't it? Having to lightly-obfuscate a name to allow for different mentions. I like what you have done with pgrep, and why!

Nachtzuster, 2024-04-28:
@tomtom215 Re-opening so I remember to look into it

lloydbayley, 2024-04-29:
I've had a think about this and I reckon it should go in the views.php in the IF statement before it goes to the phpsysinfo/index.php.
However, it should check the file first to see if it's already done it so we're not writing to that file every time we go into PHPSysinfo.

Of course, the script would ideally be changed to PHP...
Thoughts? Or has @Nachtzuster already fixed it? (I should learn to check before I type!) :)

lloydbayley, 2024-04-29:
Ahh, so it will also do it when the user updates to fix current installations. See, I knew I shouldn't be up at 2am thinking, instead of going to bed! :) Nice one!

closed 2024-05-13 by Nachtzuster