Nachtzuster/BirdNET-Pi #54: Localization setting change leads to incorrect unique detected species count

programmdesign, 2024-05-05:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
If one changes Database language in the Localization settings, previously species are double counted with their common names in several languages. E.g., if I switch from English to German, I get. in the stats "Grünfink" and "European Goldfinch". Hence, the number of identified species is wrong after the settings are updated.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools > Settings > Advanced > Localization
2. Change to another language and detect birds
3. Got to Daily stats
4. See error

**Expected behavior**
Localization should only impact displayed names, not database stats or number of unique species.

**Screenshots**
[image]
[image]

**Additional context**
-

**Your build**
- Irrelevant for this bug

**Code or log snippets**
-

Nachtzuster, 2024-05-08:
Hi, yes that would mess up the stats. 
Though it would be nice if only the ui's language would change.
The way the original authors implemented it, one should not change the language once everything is set up. (the localized common name is quite fundamental to the internal data structure)

This should probably be made clearer in the 'Settings'

arne1921KF, 2024-05-09:
This is a fundamental and very common problem of taxonomy - also for scientific names. A database of biological entities (taxa) would need to be implemented to change that behaviour, i.e. usage of a taxonomic backbone. 

I doubt I find the time to fix this, but surely there are already working solutions for that. 

I would bet there is a RESTful API for birds somewhere which could provide a current taxonomy based on IDs, and without looking at it I bet you a beverage that @kahst can provide a solution basically out of the box. 😄 

The basic idea is that one uses a stable ID instead of a name, and has relations behind that ID to deliver a) consistency in case of name changes and b), more importantly, consistency in case of merges and splits considering the ID. (Which happens, rather frequently. We even have a 'taxonomy' of taxonomists for that - we call them _splitters_ and _lumbers_ based on their favoured changes in biological taxonomy...)

Nachtzuster, 2024-05-15:
Ouch, I hope that was not too painful.

Yes, using scientific names internally is the correct thing to do for easy internationalization.
As changing this is unlikely to happen, I've added a warning in the Settings page.

closed 2025-06-29 by Nachtzuster