Nachtzuster/BirdNET-Pi #56: Issue with German location code

R3tr0BoiDX, 2024-05-10:
[image]

R3tr0BoiDX, 2024-05-10:
This is from the log online btw, in case that wasn't obvious, heh

Nachtzuster, 2024-05-12:
This seems to be an fairly common issue issue, according to Google. Not quite sure what the 'correct' fix would be, but for now, just deleting the incorrect `/etc/timezone` should 'fix' it

lloydbayley, 2024-05-16:
I've given this a great deal of thought too. It's a really tricky one. It poses the question "Can a 'complete fix' be implemented?" There are so many ways it could be 'corrupted' by the insertion of multiple timezones that it's really down to the user to control. It's hard for BirdNet-Pi to be responsible for the OS settings unless it has the audacity to check and override them, which is a big thing. On one hand, I feel that the OS settings are the user responsibility and on the other hand, I'm wondering if there might be a check by BirdNet-Pi to alert the user to the potential problem. The underlying question is, how far does the responsibilty stretch?

Nachtzuster, 2024-06-16:
@R3tr0BoiDX  I pushed a fix for this, setting the timezone should be fine now

closed 2024-06-16 by Nachtzuster

R3tr0BoiDX, 2024-06-20:
> @R3tr0BoiDX I pushed a fix for this, setting the timezone should be fine now

awesome, thanks!