Nachtzuster/BirdNET-Pi #58: $listenurl link in Apprise notification doesn't work

ghvanderweg, 2024-05-13:
I have an Apprise notification setup to a Discord webhook:

[image]

The notification itself works fine:

[image]

But when I click on the link, a PHP error is thrown:

`Fatal error: Uncaught TypeError: in_array(): Argument #2 ($haystack) must be of type array, null given in /home/gerbrand/BirdNET-Pi/scripts/play.php:527 Stack trace: #0 /home/gerbrand/BirdNET-Pi/scripts/play.php(527): in_array() #1 /home/gerbrand/BirdNET-Pi/homepage/views.php(146): include('...') #2 {main} thrown in /home/gerbrand/BirdNET-Pi/scripts/play.php on line 527`

I'm running on a Raspberry 4B with the latest OS Lite (bookworm)

ghvanderweg, 2024-05-13:
I think the issue is here: https://github.com/Nachtzuster/BirdNET-Pi/blob/43fcc09f4f72166357ab3ccf63ea7cb9a5a25cee/scripts/play.php#L526C1-L530C72

`$disk_check_exclude_arr` is null in my case, because I don't have the file `/BirdNET-Pi/scripts/disk_check_exclude.txt`.
`$disk_check_exclude_arr` should be declared as an empty array like here: https://github.com/Nachtzuster/BirdNET-Pi/blob/43fcc09f4f72166357ab3ccf63ea7cb9a5a25cee/scripts/play.php#L398

Nachtzuster, 2024-05-13:
Thanks for reporting!
`$disk_check_exclude_arr` should indeed be properly initialised

Not having a disk_check_exclude.txt is unexpected though: `~/BirdNET-Pi/scripts/disk_check_exclude.txt` gets created/updated every time the 'Best Recordings' page is visited, so I wonder if there is an additional issue here.

Could you click the 'Best Recordings' menu and see it the issue persists?

ghvanderweg, 2024-05-13:
After visiting the 'Best Recordings' page the issue does go away. It was a brand new installation, so I hadn't visited that page yet :)

lloydbayley, 2024-05-14:
Perhaps it should be created as a blank file as part of the initial install and snippets updates. Would that be easiest?

Nachtzuster, 2024-05-15:
fixed here https://github.com/Nachtzuster/BirdNET-Pi/commit/6e6e9eab089ff908af777d0875d89e79fe76a6c2
@ghvanderweg : thanks!

closed 2024-05-15 by Nachtzuster