Nachtzuster/BirdNET-Pi #67: BirdWeather "email Tim" registration method is out of date

scruss, 2024-05-18:
**Describe the bug**
You can get a BirdWeather ID yourself, and you no longer need to send email to make it happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Tools", "BirdWeather", and click on *Email Tim to request a BirdWeather ID*
2. Compose and send email
3. wait …
4. wait …
5. Eventually get an email that says *You can now use our automated form to create your BirdWeather ID."

Please update the registration method. Full text of the email follows:

> Hi!
> 
> You can now use our automated form to create your BirdWeather ID.
> 
> You’ll first need to create an account - https://app.birdweather.com/login
> 
> Once that’s done - you can go to - https://app.birdweather.com/account/stations
> 
> Make sure that the Latitude and Longitude match what is in your BirdNET-Pi configuration.
> 
> NOTE - by using your BirdWeather ID - you are consenting to sharing your soundscapes and detections with BirdWeather.
> 
> Cheers!
> 
> -Tim
> 
> Founder - BirdWeather

lloydbayley, 2024-05-20:
#68

Nachtzuster, 2024-05-21:
@scruss , @lloydbayley  merged just now, Thanks!

closed 2024-05-21 by Nachtzuster