Nachtzuster/BirdNET-Pi #70: Apprise not using Flickr account restriction

killervette42, 2024-05-19:
I set my Flickr api and account. The website pulls images from my Flickr, however the apprise notifications choose any image from Flickr. I would expect it to pick images from my account specified.

lloydbayley, 2024-05-20:
No, It's more simple than that. It uses your details to log in to Flickr and use the API. The images are pulled from Flickr based on the names.

killervette42, 2024-05-20:
Is that just for the apprise section. The webpage works correctly and shows only images from my account which is specified.

lloydbayley, 2024-05-20:
Really? Well actually, I hadn't delved that deep into it. I just accepted that it showed pictures. I have a basic Flickr account with no pictures, so I hadn't noticed! I'll have another look through the code.

killervette42, 2024-05-20:
Yea. On the webpage of my account doesn’t have a picture of a certain bird, no picture shows. But it worked perfectly for the birds I do have pics for. 

Just apprise ignores it.

lloydbayley, 2024-05-20:
You're absolutely correct. Notifications don't honour the filter. We can stick it on the to-do list! (Unless @Nachtzuster has already begun)...

killervette42, 2024-05-20:
Great, thanks!

Nachtzuster, 2024-05-21:
> stick it on the to-do list!

indeed

killervette42, 2024-05-23:
Also, it's ignored when you go to Best Recordings and then pick a species.

lloydbayley, 2024-07-12:
@Nachtzuster Did anyone look into this or is it still on 'the list' ? :)

lloydbayley, 2024-07-12:
Ok, will try to remember to have a look at this over the weekend!

lloydbayley, 2024-07-16:
Just as a small update, I've had a good look at this and it's a bit yukky....since the notify is in python and the main one is in PHP, it will require some effort. I'm going to have a think about this.

Nachtzuster, 2024-07-17:
I see. 
Maybe another option would be to expose the php flickr functionality as a (rest?) api, so that the python code can call it.
Only one place with flickr logic would be nice.
Just an idea, I have not looked into how involved that would be

lloydbayley, 2024-07-18:
It certainly makes sense. A bit beyond my skillset at the moment I'm afraid.

Nachtzuster, 2025-08-27:
@killervette42 This should be fixed now