Nachtzuster/BirdNET-Pi #81: Changing bird species gives error

Hiemstra87, 2024-06-01:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
When i try to change the species it throws an error. (See screenshot). "The first argument should be a starting with the common name of the bird and finish with mp3!, Instead, it is : Porseleinhoen***.wav<br>"
It does not change the species of the recording.

**To Reproduce**
Go to recordings -> by species -> -bird-
Tap bird icon
Choose new species
Confirm the "are you sure"
Error
Steps to reproduce the behavior:

**Expected behavior**
no error and change of species.

**Screenshots**
[image]

**Additional context**
actions from a Google Pixel phone on Chrome browser:
version 125.0.6422.146
com.android.chrome

**Your build**
Birdnet-PI build on RasPI 5, additional install of Cloudflared.

**Code or log snippets**
"The first argument should be a starting with the common name of the bird and finish with mp3!, Instead, it is : Porseleinhoen***.wav<br>"

</details>

alexbelgium, 2024-06-01:
Thanks for reporting! Indeed i considered only files saved as .mp3, I didn't knew it was possible for other formats. I'll make an update proposal to the code

Hiemstra87, 2024-06-01:
Wow that's a quick reply! Thanks!

alexbelgium, 2024-06-01:
Well there was a lack of goals in the Champions league final so I was a bit idle - this has changed however since ;)

Hiemstra87, 2024-06-01:
Yeah, bit of a disappointment Dortmund didn't win though

alexbelgium, 2024-06-02:
Indeed... Rm has enough trophies on their shelves... And I had the luxury of being perfectly neutral in this final

lloydbayley, 2024-06-02:
Nice bug-squash! I must admit, I, too, had forgotten about the possibility of changing format. I never really occurred to me to alter it but I guess when you have ffmpeg at your disposal, the world is your oyster! :)
Bonne Chance & Viel Glück to you both!

Hiemstra87, 2024-06-02:
Nice quick fix, just tested it, works like a charm. You're making a lot of people happy with this feature.

closed 2024-06-02 by Hiemstra87

lloydbayley, 2024-06-02:
He can't help it....He's a people-pleaser....