Nachtzuster/BirdNET-Pi #84: Caddy v2.8.2 breaks current and new installations

craigusus, 2024-06-02:
**Describe the bug**
I've been having stability issues with my Pi5 install, so I tried an apt update/upgrade as I know some of my other Pis have had kernel/pi firmware updates. One of the updates was to caddy. After the update I had no access to the web interface of my BirdNET-Pi. But the Pi4 I have on test in my shed, is working fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Log in via ssh
2. Run `sudo apt update && sudo apt full-upgrade -y`

Also
1. Clean install from scratch
2. Run `sudo apt update && sudo apt full-upgrade -y`
3. Run BirdNET-Pi install script

**Expected behavior**
Caddy service to not fail

**Screenshots**
Fresh install Pi5 (broken):
[image]
Test Pi4 (built yesterday, 1st June):
[image]

**Your build**
Pi5 8GB and Pi4 1GB

Aixsponsa, 2024-06-02:
I do not have anything to add besides that my web interface is also inaccessible today.

Swiftnesses, 2024-06-02:
This is fixed, as 2.8.4 has been released. Rerunning the install should do it.

craigusus, 2024-06-02:
@Swiftnesses Yes, this seems to have fixed it now. Thanks for the update.

closed 2024-06-02 by craigusus