Nachtzuster/BirdNET-Pi #92: Restore 'Update Settings' button after succes

Nachtzuster, 2024-06-08:
The `Update Settings` text changes to `Success!` after updating the settings, which is good for user feedback.
It would be nice if it changes back to `Update Settings` after a while (5s?)

alexbelgium, 2024-06-09:
Wouldn't it be better for user feedback to have a messagebox instead? and always leave the message to "Update Settings" ?

lloydbayley, 2024-06-09:
've been meaning to do something about that for quite some time and keep forgetting. No-doubt @alexbelgium has already started coding it! Not quite sure how much my input is wanted anymore but I'll check on things now and then

alexbelgium, 2024-06-09:
As a matter of fact I haven't ;) you are more than welcome to try ! Anyway the best is that the one who does it starts a branch linked to this issue so we know and avoid duplication of work

alexbelgium, 2024-06-10:
Here is a proposed code that shows a js alert message while keeping the button text constant... Not super optimized but should be working :-)

One day we might wonder why updating settings is sooooo slow and if we can optimize it too ;-)

Nachtzuster, 2024-06-10:
> sooooo slow

I think it restart services, though I guess not all settings require the restart. the code errs on the safe but slow side

closed 2024-06-16 by Nachtzuster