Nachtzuster/BirdNET-Pi #93: Issue with Unsupported Characters in Daily Charts and Spectrograms

isnkys, 2024-06-09:
Thank you for the incredible work on this project. 
While setting up the system in Japanese via Tools->Settings->Localization, I noticed that the species names in the Daily charts appeared as unsupported characters.

To address this, I installed the Noto Sans font from [here](https://fonts.google.com/noto/specimen/Noto+Sans+JP) into `homepage/static`. Additionally, I replaced `Roboto Flex` with `Noto Sans CJK JP` at line 158 in `scripts/daily_plot.py`, which successfully resolved the display issue for the Daily charts. However, the species names at the top of the spectrograms still appear as question marks. It seems this issue stems from the `get_safe_title(title)` function in `scripts/utils/reporting.py` at line 20, which converts UTF-8 characters to ASCII. I would appreciate if you could investigate and provide a solution for this.

Default font (Roboto Flex)
[image]
Font changed to Noto Sans
[image]
Spectrograms
[image]

Nachtzuster, 2024-06-10:
Thanks for reporting.

So there seem to be two separate but related issues:
- daily_plot.py: should be fixable, thanks for the details
- spectrograms: this is harder: get_safe_title() existes because `sox`, the tool that creates the spectrogram does not support non-ascii text, see https://sourceforge.net/p/sox/feature-requests/214/

Nachtzuster, 2024-06-10:
> I installed the Noto Sans font from [here](https://fonts.google.com/noto/specimen/Noto+Sans+JP)

font noob question: Is there a particular reason to pick this specific font?

kyjibato, 2024-06-11:
Thank you for your reply. The reason I chose Noto Sans is because it supports CJK (Chinese, Korean, and Japanese) characters, has a clear licensing structure, and is created by Google, similar to Roboto.
https://github.com/notofonts/noto-cjk/tree/main/Sans

Nachtzuster, 2024-06-15:
@kyjibato could you update to see if it works for you now?

Nachtzuster, 2024-06-15:
@kyjibato could you open another issue for the spectrograms?
we might have to re-implement that with python/mathplotlib

kyjibato, 2024-06-15:
The recent fix you implemented for the issue has worked perfectly in displaying Japanese characters. I truly appreciate your quick and effective response. I will open another issue.
[image]

Nachtzuster, 2024-06-15:
@kyjibato so you have a BirdNET-Pi instance actually running in Japan?

kyjibato, 2024-06-15:
@Nachtzuster Yes, I run my BirdNET-Pi in Japan. This is my station https://app.birdweather.com/stations/4987

closed 2024-06-16 by Nachtzuster