Nachtzuster/BirdNET-Pi #94: BUG - Species List Tester doesnt work anymore -> thisrun.txt got removed

MueJosh, 2024-06-10:
**Describe the bug**
Fresh Install -> Species List Tester does not work because "thisrun.txt" does not exist anymore (removed in a update)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Tools" -> "Settings" -> "Model"
2. Click on "Species List Tester" 
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen. - You know what to expext - it works on all of my old installs 
[image]

**Your build**
Pi 02W, 2GiB Swap

**Code or log snippets**
```
Traceback (most recent call last):
  File "/home/USERNAME/BirdNET-Pi/scripts/species.py", line 100, in 
    lat = str(str(str([i for i in this_run if i.startswith('LATITUDE')]).split('=')[1]).split('\\')[0])
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```
[image]

Nachtzuster, 2024-06-10:
thanks for reporting!
I must have had a bad day when I did that cleanup.

I pushed a fix, could you try?

MueJosh, 2024-06-16:
> thanks for reporting!
> 
> I must have had a bad day when I did that cleanup.
> 
> 
> 
> I pushed a fix, could you try?

Thanks! 

It works again 👍

closed 2024-06-16 by MueJosh