Nachtzuster/BirdNET-Pi #97: Feat proposal : apprise for alerts ?

alexbelgium, 2024-06-12:
Proposal : use apprise to send an alert if services are failing

Implementation : cron job to periodically monitor if "birdnet_analysis" service is active, and send a telegram alert if not ? should it also monitor recording (although recording can be paused to accomodate backlogs)

Usercase : in case the service stops for whatever reason, the user is alerted

Nachtzuster, 2024-06-15:
not sure if it would scratch your itch, but there is a heartbeat: https://github.com/Nachtzuster/BirdNET-Pi/blob/0a09e73549c0b0d4f5effbeb7088a0fcf7eb36fa/scripts/install_config.sh#L246
```
## HEARTBEAT_URL is a location to ping every time some analysis is done
## no information is sent to the the URL, its a heart beat to show that the
## analysis is continuing

HEARTBEAT_URL=
```

alexbelgium, 2024-06-15:
Thanks! I hadn't noticed it. Then indeed it is probably better to use an existing feature rather than complexifying things.

closed 2024-06-15 by alexbelgium