Nachtzuster/BirdNET-Pi #104: Issue with Unsupported Characters in Spectrograms

kyjibato, 2024-06-15:
While setting up the system in Japanese via Tools -> Settings -> Localization, I noticed that the species names in the spectrograms appeared as unsupported characters.
It seems this issue stems from the get_safe_title(title) function in scripts/utils/reporting.py at line 20, which converts UTF-8 characters to ASCII.
This problem is due to limitations in the tool `sox`. https://github.com/Nachtzuster/BirdNET-Pi/issues/93#issuecomment-2159128419
Thank you.
[image]

Nachtzuster, 2025-05-11:
@kyjibato A change was pushed to fix this.

kyjibato, 2025-05-11:
Thank you for your hard work. I'll check it soon.

closed 2025-06-18 by kyjibato