Nachtzuster/BirdNET-Pi #108: Feature to ignore region for a select list of birds

dalogue1, 2024-06-17:
I would like to request a feature to ignore the region restriction for accepting a particular bird species.

Currently I have a system running and have identified many types of birds.  However, the bird that is by far the most frequent visitor in my backyard is a type of bird that is not supposed to be in the Americas at all.  (Scaly Breasted Munia which is native to southern Asia.). The only way to get the system to add the detections to the results is to turn the threshhold down to 0.0005 because the Scaly Breasted Munias number is 0.00055 in Houston, Texas.

Lowering the threshhold does include the munias in the results, but also causes many false identifications of other birds from around the world as well.

I would like to be able to accept a bird as a possibility based upon the region model OR if it is on a list.

Note: Before anyone asks, I am absolutely certain the birds are Scaly Breasted Munias as they are always at my bird feeder and I can see as many as 10-12 at a time.  I have photos of the birds as proof.  The flock of the birds live in the tree in my backyard.

alexbelgium, 2024-06-17:
As it took me just a couple minutes I made a PR to just see if it is useful ; you could even modify the files manually to have this feature on your system ;-)

dalogue1, 2024-06-17:
Sorry, but what is a PR?

alexbelgium, 2024-06-17:
Version I've just pushed now has a workable web UI

dalogue1, 2024-06-17:
Is there an example format somewhere?  I guess I can temporarily add the Scaly Breasted Munia to the Exclude list and then copy the Exclude list file to be the Whitelist file then remove it from the Exclude List.

dalogue1, 2024-06-17:
the singular form of species is also species.   (not specie)

Sorry to be pedantic.

alexbelgium, 2024-06-17:
> Is there an example format somewhere? I guess I can temporarily add the Scaly Breasted Munia to the Exclude list and then copy the Exclude list file to be the Whitelist file then remove it from the Exclude List.

You can do that, or just use the webui I've made - you can simply open the current files and replace the content with the code from the PR.

Thanks for the "species" precision, I was not aware!

dalogue1, 2024-06-17:
I am curious that in python the clause 

       "(entry[0] in PREDICTED_SPECIES_LIST or WHITELIST_LIST or len(PREDICTED_SPECIES_LIST) == 0)"

treats the WHITELIST_LIST  as part of the 'in' statement.

I would have expected the following  

       "(entry[0] in PREDICTED_SPECIES_LIST or entry[0] in WHITELIST_LIST or len(PREDICTED_SPECIES_LIST) == 0)"

alexbelgium, 2024-06-17:
hehe you are very probably right :-) I'll adapt the python I went way too fast and don't have a usecase to actually test it

dalogue1, 2024-06-17:
No problem.  Thank you for the quick enhancement.

Nachtzuster, 2024-06-17:
@dalogue1 does using the v1 species range model help?
(ie. uncheck the V2 model)
[image]

dalogue1, 2024-06-17:
no both Models (with and without The Species range model switch)  see the Scaly Breasted Munias but fail to "Detect" them.

Look at the following image of the expected locations of that type of bird.  I find it amusing that I have so many of these birds in my yard.  (In fact I do not see generic house sparrows in my yard as I think the munias keep them away.)

[image]

Nachtzuster, 2024-06-17:
> They can have a recognition of over 90% and still not accept them.

interesting, could you provide a log?

dalogue1, 2024-06-17:
Sorry   but the changes to the code did not work to Whitelist birds.
### With a 0.0005 threshold

18:07:19---[server][INFO] 3.0;6.0-('Lonchura punctulata_Scaly-breasted Munia', 0.88426113)
18:07:19---[server][INFO] 6.0;9.0-('Lonchura punctulata_Scaly-breasted Munia', 0.8649536)
18:07:19---[server][INFO] 9.0;12.0-('Lonchura punctulata_Scaly-breasted Munia', 0.9807278)
18:07:19---[server][INFO] 12.0;15.0-('Lonchura punctulata_Scaly-breasted Munia', 0.763961)
### 18:07:20---[birdnet_analysis][INFO] 2024-06-17;18:07:04;Lonchura punctulata;Scaly-breasted Munia;0.9807;29.7349;-95.4698;0.7;25;1.25;0.0;Scaly-breasted_Munia-98-2024-06-17-birdnet-18:07:04.mp3
### 18:07:20---[birdnet_analysis][INFO] 2024-06-17;18:07:04;Lonchura punctulata;Scaly-breasted Munia;0.9807;29.7349;-95.4698;0.7;25;1.25;0.0;Scaly-breasted_Munia-98-2024-06-17-birdnet-18:07:04.mp3
### 18:07:20---[birdnet_analysis][INFO] 2024-06-17;18:07:04;Lonchura punctulata;Scaly-breasted Munia;0.9807;29.7349;-95.4698;0.7;25;1.25;0.0;Scaly-breasted_Munia-98-2024-06-17-birdnet-18:07:04.mp3
### 18:07:20---[birdnet_analysis][INFO] 2024-06-17;18:07:04;Lonchura punctulata;Scaly-breasted Munia;0.865;29.7349;-95.4698;0.7;25;1.25;0.0;Scaly-breasted_Munia-86-2024-06-17-birdnet-18:07:04.mp3

### With threshold of 0.001

18:09:03---[server][INFO] 0.0;3.0-('Sitta canadensis_Red-breasted Nuthatch', 0.11696951)
18:09:03---[server][INFO] 3.0;6.0-('Coccothraustes vespertinus_Evening Grosbeak', 0.1392667)
18:09:03---[server][INFO] 6.0;9.0-('Lonchura punctulata_Scaly-breasted Munia', 0.943475)
18:09:03---[server][INFO] 9.0;12.0-('Lonchura punctulata_Scaly-breasted Munia', 0.82163304)
18:09:03---[server][INFO] 12.0;15.0-('Chordeiles minor_Common Nighthawk', 0.34993884)
18:09:16---[birdnet_analysis][INFO] Analyzing BirdSongs/StreamData/2024-06-17-birdnet-18:09:01.wav
18:09:16---[server][INFO] READING AUDIO DATA...
18:09:16---[server][INFO] READING DONE! READ 5 CHUNKS.
18:09:16---[server][INFO] ANALYZING AUDIO...
18:09:16---[server][INFO] DONE! Time 0.51 SECONDS
### 18:09:16---[server][INFO] 0.0;3.0-('Lonchura punctulata_Scaly-breasted Munia', 0.121780016)
18:09:16---[server][INFO] 3.0;6.0-('Engine_Engine', 0.14332145)
18:09:16---[server][INFO] 6.0;9.0-('Human_Human', 0.0)
18:09:16---[server][INFO] 9.0;12.0-('Human_Human', 0.0)
18:09:16---[server][INFO] 12.0;15.0-('Human_Human', 0.0)

dalogue1, 2024-06-17:
Give me a version of server.py with debug prints and I will install it.

alexbelgium, 2024-06-18:
what is the content of your file `$HOME/BirdNET-Pi/scripts/whitelist_species_list.txt`? It should be `Lonchura punctulata_Scaly-breasted Munia` ; and you could try to chmod the permissions to pi:pi also

I've thought of a better way to include the option in the Pull Request : we could append the content of the whitelist_species_list.txt to the list generated by the threshold method - this would ensure the function without adding any overhead to the analysis logic

dalogue1, 2024-06-21:
There was a problem in the **server.py** code modificationof predictSpeciesList.  

The item to append was **species** not **species[0]**.  

As it was, only the letter 'L' was being added to the list from the 'Lonchura punctulata_Scaly-Breasted Munia' whitelisted bird.

    WHITELIST_LIST = None
    WHITELIST_LIST = loadCustomSpeciesList(os.path.expanduser("~/BirdNET-Pi/whitelist_species_list.txt"))
    l_filter = explore(lat, lon, week)
    for s in l_filter:
        if s[0] >= float(sf_thresh):
            # if there's a custom user-made include list, we only want to use the species in that
            if (len(INCLUDE_LIST) == 0):
                PREDICTED_SPECIES_LIST.append(s[1])
    if (len(WHITELIST_LIST) != 0):
        for species in WHITELIST_LIST:
            PREDICTED_SPECIES_LIST.append(species)

dalogue1, 2024-06-21:
It did not add the bird, but I do not know why.   I will continue to add debug prints to find out why.

> 8:24---[server][INFO] READING AUDIO DATA...
> 18:38:24---[server][INFO] READING DONE! READ 5 CHUNKS.
> 18:38:24---[server][INFO] ANALYZING AUDIO...
> 18:38:24---[server][INFO] DONE! Time 0.50 SECONDS
> 18:38:24---[server][INFO] 0.0;3.0-('Human_Human', 0.0)
> 18:38:24---[server][INFO] 3.0;6.0-('Human_Human', 0.0)
**> 18:38:24---[server][INFO] 6.0;9.0-('Lonchura punctulata_Scaly-breasted Munia', 0.8337214)**
> 18:38:24---[server][INFO] 9.0;12.0-('Empidonax flaviventris_Yellow-bellied Flycatcher', 0.20565923)
> 18:38:24---[server][INFO] 12.0;15.0-('Human_Human', 0.0)
> 18:38:39---[birdnet_analysis][INFO] Analyzing BirdSongs/StreamData/2024-06-21-birdnet-18:38:24.wav
> 18:38:39---[server][INFO] READING AUDIO DATA...
> 18:38:39---[server][INFO] READING DONE! READ 5 CHUNKS.
> 18:38:39---[server][INFO] ANALYZING AUDIO...
> 18:38:39---[server][INFO] DONE! Time 0.51 SECONDS
> 18:38:39---[server][INFO] 0.0;3.0-('Human_Human', 0.0)
> 18:38:39---[server][INFO] 3.0;6.0-('Human_Human', 0.0)
> 18:38:39---[server][INFO] 6.0;9.0-('Spinus tristis_American Goldfinch', 0.13049889)
> 18:38:39---[server][INFO] 9.0;12.0-('Haemorhous mexicanus_House Finch', 0.108848475)
> 18:38:39---[server][INFO] 12.0;15.0-('Human_Human', 0.0)

alexbelgium, 2024-06-22:
Thanks, I've implemented the corrected Py in the PR. Indeed perhaps it is due to human detection

dalogue1, 2024-06-24:
> Thanks, I've implemented the corrected Py in the PR. Indeed perhaps it is due to human detection

I am still checking the change.  I made a simple change to the code to capture a wav file that I can submit at will to see if the change works.

dalogue1, 2024-06-25:
I did a fresh install of PI OS 64-bit (bookworm) and BirdNET-Pi (Nachtzuster) from GitHub as well.  Since your changes are not in the official package, I just added the following line at the end of the **predictSpeciesList** method in **server.py**:
>             PREDICTED_SPECIES_LIST.append('Lonchura punctulata_Scaly-breasted Munia')

I then copied a saved wav file with a 97.2% confidence of a Scaly-breasted Munia into the ~/BirdSongs/StreamData directory.  The server immediately saved the bird in my detections history.  When the change is committed I will get the code to edit a whitelist with a file automatically.  (Sorry I do not know how to merge your changes into the base code without manually updating each file by hand.)

(Note: I have attached the file for the Munia as well as a Blue-Jay, a Northern Cardinal, and a White-Winged Dove.  Unzip the files and change the extension to .wav by removing the bird name. These are very handy for testing.)
[SAVED WAV FILES.zip](https://github.com/user-attachments/files/15963953/SAVED.WAV.FILES.zip)

dalogue1, 2024-06-25:
BTW I am using the V2.4 Model with the Species range model unchecked and a Threshold of 0.05

alexbelgium, 2024-06-25:
> I did a fresh install of PI OS 64-bit (bookworm) and BirdNET-Pi (Nachtzuster) from GitHub as well. Since your changes are not in the official package, I just added the following line at the end of the **predictSpeciesList** method in **server.py**:
> 
> > ```
> >         PREDICTED_SPECIES_LIST.append('Lonchura punctulata_Scaly-breasted Munia')
> > ```
> 
> I then copied a saved wav file with a 97.2% confidence of a Scaly-breasted Munia into the ~/BirdSongs/StreamData directory. The server immediately saved the bird in my detections history. When your changes are committed I will update my server with your code to edit a whitelist with a file automatically, but this proves that the changes work. I had previously tested your code for editing the whitelist file and the other changes. (Sorry I do not know how to automatically merge your changes into the base code without manually updating each file and the changes require merging of the files.)
> 
> (Note: I have attached the file for the Munia as well as a Blue-Jay, a Northern Cardinal, and a White-Winged Dove. Unzip the files and change the extension to .wav by removing the bird name. These are very handy for testing.) [SAVED WAV FILES.zip](https://github.com/user-attachments/files/15963953/SAVED.WAV.FILES.zip)

Thanks! I've synced the branch so you can just copy paste!

dalogue1, 2024-06-26:
I put the code in place and the Whitelist does not work.  I go back to the original code and add the one species in the server.py code as I described above with a single line to append the name of the bird and it works,

I will still continue to try to figure out why it is not working but I have not figured it out yet.

alexbelgium, 2024-07-01:
Hi, the PR is merged, I think we can close this. I had checked that with the current code the content of the whitelist file is appended to the detections list

dalogue1, 2024-07-03:
I finally checked the whitelist function and it works so this issue can be closed.

closed 2024-07-03 by dalogue1