Nachtzuster/BirdNET-Pi #112: Configure RTSP stream fails

bdeschut, 2024-06-17:
When going to Tools > Settings > Advanced settings
 + configuring an RTSP steam
 + pressing save at the bottom of the page
I get a pop-up saying "Settings successfully updated". However, the field where I entered the RTSP stream info is empty again and the "live audio" confirm that the local mircrophone is still used. birdnet.conf setting "RTSP_STREAM" is empty as well.

**Additional context**
1/ I have configured this around december 2023 / januari 2024 using this repo and it worked back then
2/ When configuring the RTSP stream directly in birdnet.conf and restarting core services, it works like a charm. The RTSP setting is also visible in the advanced menu
3/ Going to the advanced settings and pressing save, clears the field again. The setting is also removed in birdnet.conf

**Your build**
PI5 + bookworm

I don't see any relevant errors using "journalctl -f" and saving the RTSP stream setting.
If I find the time I will troubleshoot myself, but it might take a while

lloydbayley, 2024-06-18:
Absolutely correct. Will investigate!

lloydbayley, 2024-06-18:
Holy crap. It's too late at night for me to cope with that code. Jesus. I'll have another look tomorrow. I may even have to locally reformat it first to debug it. It's BAD!

Nachtzuster, 2024-06-18:
@bdeschut seems 'someone' screwed up... could you try now :blush:

bdeschut, 2024-06-18:
@Nachtzuster Can confirm it works now, thanks!

closed 2024-06-18 by bdeschut

lloydbayley, 2024-06-20:
How the hell did you find that in that code! I KNOW you didn't write that!

Nachtzuster, 2024-06-21:
> How the hell did you find that in that code! I KNOW you didn't write that!

@lloydbayley no, but I did break it just last week :blush:

lloydbayley, 2024-06-24:
Leuk!