Nachtzuster/BirdNET-Pi #114: Spectrogram fails to load stream

baloor666, 2024-06-19:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Spectrogram fails to load stream, just shows static spectrogram image

**To Reproduce**
Steps to reproduce the behavior:
1. Go to my BirdPi homepage
2. Click on 'Spectrogram'
3. Spectrogram page displays loading....
4. Instead of the rtsp stream, a static spectrogram image is displayed.

**Expected behavior**
Load the rtsp stream and display the animated feed.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
rpi 4b 8gb
bullseye with latest updates.
birdnet-pi current

**Code or log snippets**
logs show this error repeating every few seconds

Jun 19 14:13:26 birdpi livestream.sh[1859]: [tcp @ 0x55946dd6e0] Connection to tcp://localhost:8000 failed: Connection refused
Jun 19 14:13:26 birdpi livestream.sh[1859]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
Jun 19 14:13:26 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
Jun 19 14:13:26 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.

Jun 19 14:25:13 birdpi livestream.sh[3350]: [tcp @ 0x559629f6e0] Connection to tcp://localhost:8000 failed: Connection refused
Jun 19 14:25:13 birdpi livestream.sh[3350]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
Jun 19 14:25:13 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
Jun 19 14:25:13 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
Jun 19 14:25:16 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 401.
Jun 19 14:25:16 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
Jun 19 14:25:16 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
Jun 19 14:25:16 birdpi livestream.sh[3353]: [tcp @ 0x55a60956e0] Connection to tcp://localhost:8000 failed: Connection refused
Jun 19 14:25:16 birdpi livestream.sh[3353]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
Jun 19 14:25:16 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
Jun 19 14:25:16 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
Jun 19 14:25:19 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 402.
Jun 19 14:25:19 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
Jun 19 14:25:19 birdpi systemd[1]: Started BirdNET-Pi Live Stream.

lloydbayley, 2024-06-20:
Good question. It's never been animated. It's a snapshot of exactly what it's hearing the time when you click Spectrogram.

Nachtzuster, 2024-06-20:
@lloydbayley it is on my machine! :clown_face: 
the snapshot is a fallback for when 
- the user is not logged in 
- or the stream is not working or disabled

In this case, I guess it is safe to say the livestream is not working :smiley: 

@baloor666 what does `sudo journalctl --no-pager -u livestream.service` and `sudo journalctl --no-pager -eu livestream.service` return?

also, does restarting the livestream help? in the menu: Tools->Services->Live Audio Stream->Restart
could you post what that returns too?

baloor666, 2024-06-20:
sudo journalctl --no-pager -u livestream.service

May 07 02:31:54 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:31:54 birdpi livestream.sh[1148138]: [tcp @ 0x556e2146e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:31:54 birdpi livestream.sh[1148138]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:31:54 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:31:54 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:31:57 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112796.
May 07 02:31:57 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:31:57 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:31:57 birdpi livestream.sh[1148168]: [tcp @ 0x55a89566e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:31:57 birdpi livestream.sh[1148168]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:31:57 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:31:57 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:00 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112797.
May 07 02:32:00 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:32:01 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:32:01 birdpi livestream.sh[1148211]: [tcp @ 0x5589e736e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:32:01 birdpi livestream.sh[1148211]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:32:01 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:32:01 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:04 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112798.
May 07 02:32:04 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:32:04 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:32:04 birdpi livestream.sh[1148248]: [tcp @ 0x55866b66e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:32:04 birdpi livestream.sh[1148248]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:32:04 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:32:04 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:07 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112799.
May 07 02:32:07 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:32:08 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:32:08 birdpi livestream.sh[1148328]: [tcp @ 0x558ae616e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:32:08 birdpi livestream.sh[1148328]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:32:08 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:32:08 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:11 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112800.
May 07 02:32:11 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:32:11 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:32:11 birdpi livestream.sh[1148358]: [tcp @ 0x55794846e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:32:11 birdpi livestream.sh[1148358]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:32:11 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:32:11 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:14 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112801.
May 07 02:32:14 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:32:14 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:32:15 birdpi livestream.sh[1148389]: [tcp @ 0x558e53a6e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:32:15 birdpi livestream.sh[1148389]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:32:15 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:32:15 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:18 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112802.
May 07 02:32:18 birdpi systemd[1]: Stopped BirdNET-Pi Live Stream.
May 07 02:32:18 birdpi systemd[1]: Started BirdNET-Pi Live Stream.
May 07 02:32:18 birdpi livestream.sh[1148435]: [tcp @ 0x55b6c886e0] Connection to tcp://localhost:8000 failed: Connection refused
May 07 02:32:18 birdpi livestream.sh[1148435]: icecast://source:birdnetpi@localhost:8000/stream: Connection refused
May 07 02:32:19 birdpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
May 07 02:32:19 birdpi systemd[1]: livestream.service: Failed with result 'exit-code'.
May 07 02:32:22 birdpi systemd[1]: livestream.service: Scheduled restart job, restart counter is at 112803.

sudo journalctl --no-pager -eu livestream.service

[... cut ...]
Jun 21 08:59:54 birdpi icecast2[212666]: [2024-06-21  08:59:54] WARN CONFIG/_parse_root Warning,  not configured, using default value "localhost". This will cause problems, e.g. with YP directory listings.
Jun 21 08:59:54 birdpi icecast2[212666]: [2024-06-21  08:59:54] WARN CONFIG/_parse_root Warning,  not configured, using default value "Earth".
Jun 21 08:59:54 birdpi icecast2[212666]: [2024-06-21  08:59:54] WARN CONFIG/_parse_root Warning,  contact not configured, using default value "icemaster@localhost".

baloor666, 2024-06-21:
This appears to have resolved itself.
Uncertain on the cause as I haven't made any modifications.

closed 2024-06-21 by baloor666

lloydbayley, 2024-06-21:
> @lloydbayley it is on my machine! 🤡 the snapshot is a fallback for when
> 
>     * the user is not logged in
> 
>     * or the stream is not working or disabled

I'm hoping that wasn't meant to be that I am a clown, however, in that case, I did further testing. I've NEVER seen it work. I had to try it in Google Chrome to find out that it does indeed animate. So all these years running it in Firefox or Safari, I've never seen it work. This may require further investigation. I can test things but I won't pursue a fix in case I make a fool of myself again.

Nachtzuster, 2024-06-21:
> I'm hoping that wasn't meant to be that I am a clown

I was not, I tried to indicate a lame joke there.
I tested this successfully on Firefox on both Win11 and Linux, so interesting that it does not work for you. Open an issue if you can't figure it out?

Nachtzuster, 2024-06-21:
> This appears to have resolved itself. Uncertain on the cause as I haven't made any modifications.

@baloor666 I suspect it might be a race condition