Nachtzuster/BirdNET-Pi #135: Incorrect Flickr URL being generated

Nachtzuster, 2024-07-02:
### Discussed in https://github.com/Nachtzuster/BirdNET-Pi/discussions/132

<div type='discussions-op-text'>

<sup>Originally posted by **eDad2003** July  1, 2024</sup>
First of all, I'm so happy to have found this repo!  It's really great that BirdNet-Pi will live on!

I was wondering if this is problem is experienced by anyone else?

I added the Flickr API yesterday, so I now have inline images associated with detections.  When the image is clicked, you get a popup window with an expanded image including 3 URLs - Image link, Author link, and License URL. The 'Image link' is always reported by Flickr as 'not found'.

By playing around a bit, I determined that the URL is generated incorrectly - the image is not at AuthorURL+ID.  See below for samples.

I traced it back to PHP in overview.php and todays_detections.php.  Here's the code I changed to correct it:
`$modaltext = $flickr_cache["author_url"] . "/" . $flickr_cache["id"];`
`//next line added by me to correct Flickr 'Image Link' URL`
`$modaltext = str_replace("people", "photos", $modaltext);`

It works... kinda.  There must be some caching or something else going on. I manually edit those PHP files (via file manager), save it, and then do a system reboot, but sometimes the incorrect version is still present. I'm kinda confused, but I'm new to this.

**Working URL - After code change**
[image]

**Incorrect URL - Before code change**
[image]
</div>

Nachtzuster, 2024-07-02:
@eDad2003 thanks for the analysis.
And yes, there is indeed caching.

On your patched system you could delete `scripts/flickr.db` and reboot to reset the cache.

Emmo213, 2024-07-03:
Pull request opened. I also added a "Clear Flickr Cache" button in the settings.

closed 2024-07-05 by Nachtzuster