Nachtzuster/BirdNET-Pi #140: Blacklisting flickr images doesn't work

AdaMacey, 2024-07-05:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
When viewing the overview page or Today's detections, an image of each bird is pulled from flickr and displayed by the detection.  

I have an issue where the image being pulled is for the incorrect bird (this is not a bug in itself, and happens due to ambiguous tagging on the flickr image itself). I click on the image and have an option to blacklist that particular image, however, the blacklisting never takes effect.  The blacklisted image continues to display for previous and new detections of the bird.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Today's detections" 
2. Click on an image of a bird
3. Click on "Blacklist this image" on a commonly detected bird
4. The image will continue to to appear for new detections 

**Expected behavior**
The image should no longer appear for new detections, and ideally, it should not be used for previous detections either.

**Your build**
[image]

Running version: [a9c8d9a235ff74f8bcaa2ec833cf4912abbab91e](https://github.com/Nachtzuster/BirdNET-Pi/commit/a9c8d9a235ff74f8bcaa2ec833cf4912abbab91e)

Nachtzuster, 2024-07-06:
Thanks for reporting, I can't reproduce this unfortunately.
Could you run `cat scripts/exclude_species_list.txt` 
and `sqlite3 -header -csv scripts/flickr.db "select * from images;"` and post the results?

AdaMacey, 2024-07-07:
Nothing from the cat (I don't have any excluded species)

From the sql extract, the specific image is for Cracticus torquatus. The bird is called a "Gray Butcherbird" but the flickr api is returned [a photo of a different type of butcherbird](https://farm5.static.flickr.com/4838/45201584935_de27c38a53.jpg), photographed by a man named Gray. 

```
birdpie@birdpie ~/BirdNET-Pi (main)> sqlite3 -header -csv scripts/flickr.db "select * from images;"
sci_name,com_en_name,image_url,title,id,author_url,license_url,date_created
"Anas platyrhynchos",Mallard,https://farm1.static.flickr.com/340/31688165684_c7ae3fccc6.jpg,"My bankside companion",31688165684,https://flickr.com/people/142252831@N04,https://creativecommons.org/licenses/by-sa/2.0/,2024-06-08
"Cracticus torquatus","Gray Butcherbird",https://farm5.static.flickr.com/4838/45201584935_de27c38a53.jpg,"Lahtora Butcher Bird (Lanius Lahtora) from Illustrations of Indian zoology (1830-1834)by John Edward Gray (1800-1875). Original from The New York Public Library. Digitally enhanced by rawpixel.",45201584935,https://flickr.com/people/153584064@N07,https://creativecommons.org/licenses/by/2.0/,2024-06-08
"Manorina flavigula","Yellow-throated Miner",https://farm8.static.flickr.com/7328/28000567241_d3f6292e97.jpg,"Yellow-throated miner",28000567241,https://flickr.com/people/64249609@N08,https://creativecommons.org/publicdomain/zero/1.0/,2024-06-09
"Tyto alba","Barn Owl",https://farm3.static.flickr.com/2390/1766018873_03a4318525.jpg,"Barn Owl",1766018873,https://flickr.com/people/40648743@N00,https://creativecommons.org/licenses/by/2.0/,2024-06-09
"Anthochaera carunculata","Red Wattlebird",https://farm2.static.flickr.com/1508/24347399189_f1f430cb48.jpg,DSC_4654,24347399189,https://flickr.com/people/51156152@N00,https://creativecommons.org/licenses/by/2.0/,2024-06-12
"Tringa nebularia","Common Greenshank",https://farm5.static.flickr.com/4894/46032622024_feab3d770f.jpg,"Common Greenshank (Tringa nebularia) & Common Sandpiper (Actitis hypoleucos)",46032622024,https://flickr.com/people/68466173@N02,https://creativecommons.org/licenses/by-sa/2.0/,2024-06-13
"Corvus coronoides","Australian Raven",https://farm4.static.flickr.com/3004/3069953593_c5df1dc2a3.jpg,"Australian Raven_0039",3069953593,https://flickr.com/people/28597232@N02,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-14
"Fulica atra","Eurasian Coot",https://farm5.static.flickr.com/4395/37460770775_4466faf304.jpg,"Eurasian Coot - Fulica atra - Barton - ACT - Australia - 20170924 @ 16:53",37460770775,https://flickr.com/people/56279172@N04,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-17
"Platycercus adscitus","Pale-headed Rosella",https://farm3.static.flickr.com/2092/31991366234_a4e53e18a1.jpg,"Pale Headed Rosella",31991366234,https://flickr.com/people/24526477@N06,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-17
"Vanellus miles","Masked Lapwing",https://farm2.static.flickr.com/1846/43951323214_3857cb5cc2.jpg,"Masked Lapwing",43951323214,https://flickr.com/people/156515825@N04,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-20
"Corvus orru","Torresian Crow",https://farm2.static.flickr.com/1623/25931439922_bbfae0401f.jpg,"Torresian Crow",25931439922,https://flickr.com/people/101736587@N04,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-25
"Trichoglossus moluccanus","Rainbow Lorikeet",https://farm9.static.flickr.com/8555/30234144845_8246129e62.jpg,"Rainbow Lorikeet (Trichoglossus moluccanus)",30234144845,https://flickr.com/people/63175631@N02,https://creativecommons.org/licenses/by/2.0/,2024-06-25
"Cacatua galerita","Sulphur-crested Cockatoo",https://farm6.static.flickr.com/5761/23979854841_ee6b62dbde.jpg,"Sulphur-crested Cockatoo",23979854841,https://flickr.com/people/73978954@N00,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-26
"Hirundo neoxena","Welcome Swallow",https://farm5.static.flickr.com/4594/24370900547_d0e9443c72.jpg,"Welcome Swallow",24370900547,https://flickr.com/people/84148232@N00,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-28
"Manorina melanocephala","Noisy Miner",https://farm3.static.flickr.com/2384/32919942162_914fe7f379.jpg,"Noisy miner (Manorina melanocephala) (25 centimetres)",32919942162,https://flickr.com/people/101736587@N04,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-28
"Dacelo novaeguineae","Laughing Kookaburra",https://farm1.static.flickr.com/963/42244402882_9283b5e7f2.jpg,"Laughing Kookaburra",42244402882,https://flickr.com/people/122495188@N08,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-29
"Numenius phaeopus",Whimbrel,https://farm2.static.flickr.com/1735/40852138340_01c87c79ce.jpg,Whimbrel,40852138340,https://flickr.com/people/25492008@N08,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-29
"Gymnorhina tibicen","Australian Magpie",https://farm5.static.flickr.com/4340/36139863160_efd9b0e245.jpg,"Gymnorhina tibicen",36139863160,https://flickr.com/people/78247220@N08,https://creativecommons.org/licenses/by-nc/2.0/,2024-06-29
"Sphecotheres vieilloti","Australasian Figbird",https://farm2.static.flickr.com/1872/30533776108_00992184b5.jpg,"Australasian Figbird (Specotheres vieilloti) (female) (25 centimetres)",30533776108,https://flickr.com/people/101736587@N04,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-06-30
[... cut ...] ons.org/licenses/by/2.0/,2024-07-01
"Strepera graculina","Pied Currawong",https://farm7.static.flickr.com/6170/6174155101_0d741a1ac2.jpg,"Pied Currawong where it should be!",6174155101,https://flickr.com/people/58847482@N03,https://creativecommons.org/licenses/by-nc-nd/2.0/,2024-07-03
"Eolophus roseicapilla",Galah,https://farm4.static.flickr.com/3735/20213975785_5924f0b732.jpg,Galahs,20213975785,https://flickr.com/people/65749227@N00,https://creativecommons.org/licenses/by-sa/2.0/,2024-07-06
```

Nachtzuster, 2024-07-07:
my bad, I meant `cat scripts/blacklisted_images.txt` not `exclude_species_list.txt`

AdaMacey, 2024-07-07:
```
birdpie@birdpie ~/BirdNET-Pi (main)> cat scripts/blacklisted_images.txt
45201584935
45201584935
45201584935
45201584935
45201584935
```

Nachtzuster, 2024-07-08:
Thanks. I see you tried to blacklist the picture five times :-)

The offending picture _is_ in the cache indeed, weird.
Sanity check: Do you have this user field configured?
[image]

Nachtzuster, 2024-07-08:
Another question: does blacklisting pictures of other species work?

AdaMacey, 2024-07-08:
I don't have the flickr account field filled out, because I don't know of a flickr account with photos of all of the birds from my area.

I hadn't testing blacklisting on other birds prior, but I just tried it now, and it worked immediately. When the page reloaded, it was using a new picture

Nachtzuster, 2024-07-09:
I've created a debug branch to try and figure out what is going on. Could you switch to that branch to gather the debug log?

To switch to the branch:
```shell
git remote remove origin
git remote add origin https://github.com/Nachtzuster/BirdNET-Pi.git
git fetch
./scripts/update_birdnet.sh -b debug_flickr
git status
```
that last command should confirm that you are indeed on the debug branch
after that reset the debug log:
```
rm -f ../BirdSongs/Extracted/debug_log.log
tail -F ../BirdSongs/Extracted/debug_log.log
```
this last command will complain about the file being missing, ignore it

Then go to the gui and try to blacklist the offending picture and post the debug log

AdaMacey, 2024-07-09:
```
birdpie@birdpie ~/BirdNET-Pi (debug_flickr)> tail -F ../BirdSongs/Extracted/debug_log.log
tail: cannot open '../BirdSongs/Extracted/debug_log.log' for reading: No such file or directory
tail: '../BirdSongs/Extracted/debug_log.log' has appeared;  following new file
45201584935 found in blacklist
https://www.flickr.com/services/rest/?method=flickr.photos.search&api_key=960b2cb31ff87e9af6050913cd5ea0b0&text=Gray%20Butcherbird%20bird&sort=relevance&license=2%2C3%2C4%2C5%2C6%2C9&orientation=square,portrait&per_page=5&media=photos&format=json&nojsoncallback=1
json count: 1
$photo=== null
set_image_in_db() Cracticus torquatus, Gray Butcherbird, https://farm.static.flickr.com//_.jpg, , , https://flickr.com/people/,
```

Nachtzuster, 2024-07-10:
Thanks!
So it seems Flickr has exactly one picture of a 'Gray Butcherbird' (which is actually a Lahtora Butcher Bird ) and therefore (current) BirdNET cannot replace it with a good one, hence you keep seeing the wrong picture.

After I push the fix you'll have no picture for the Gray Butcherbird species of course, but I guess that's better that the incorrect one.

Also, good catch on the API key ;-)

AdaMacey, 2024-07-10:
Thanks! It just so happens I have photos of Gray Butcherbirds! So I better get uploading :)

AdaMacey, 2024-07-10:
So, looking in to it, it's a bit more complex than it seems.  The RPI is return "Gray Butcherbird", but "Gray" is the Americanised spelling. In Australia, we use Grey rather than Gray. So the bird is called a Grey Butcherbird by Australians, and because it's an Australian bird, that's what you'll find on flickr.

lloydbayley, 2024-07-12:
That's a bit interesting. I'm almost positive I was getting pictures of the Grey Butcherbird (even if it is spelled the yankee way) :) I wonder if someone has removed their flickr account or pictures and therefore caused the lack of pictures?

Nachtzuster, 2024-07-12:
I is a pretty bizarre corner case to be sure, but it is totally reproducible: the 'a' version of the query returns the one wrong result, the 'e' version returns 76 results. I haven't looked at the actual pictures, but the captions are promising for page one, at least.

[image]
vs
[image]

lloydbayley, 2024-07-12:
Agreed. Looks promising. I'll keep an eye on this over the next few days. I get loads of these buggers every day shouting around the place, so I will be able to see what it is trying.

Nachtzuster, 2024-07-12:
@AdaMacey I've pushed a fix (actually two fixes; the other is semi-related) 
to update Tools->System Controls->Update

The spelling should be fixed too in order to get the 'e' version, obviously. Until I get around to do that, you can fix it locally by running `sed -i 's/Gray Butcherbird/Grey Butcherbird/' model/labels_flickr.txt`

AdaMacey, 2024-07-13:
Thanks. All updated. Lets see how it goes! Like lloyd, it's butcherbird central where I am :)

AdaMacey, 2024-07-13:
And the first test case just sang a song for me, and it's all working!  

Thank you!

closed 2024-07-16 by Nachtzuster