Nachtzuster/BirdNET-Pi #143: New Feature Request: New Species Today

wenestvedt, 2024-07-10:
In addition to the five statistics on the Overview page (namely, Total; Today; Last Hour; Species Detected Today; and Total Number of Species), can we get a report of any _new_ species spotted, please?

alexbelgium, 2024-07-12:
Interesting idea ; and by clicking on it, it could list only observations relevant to those species.

I wonder if "new" is better, or "rare" ; where "rare" would be any species not detected in the last 7 days for example. This might allow to highlight very rare birds that have already came, but not often.

For info, the weekly report provides the specific "new" species per week.

lloydbayley, 2024-07-12:
The 'new' idea could be handy for quickly checking new identifications to rule out false positives if you don't think to look at the weekly report (Which I MAY or MAY NOT be doing.....hehe)

lloydbayley, 2024-08-26:
Understood. :)

alexbelgium, 2024-10-08:
Hi, I've proposed a feature to highlight the new species in a day : https://github.com/Nachtzuster/BirdNET-Pi/pull/213

I use it most to sort false positives, in addition to the telegram notification on new species

wenestvedt, 2025-01-03:
> Hi, I've proposed a feature to highlight the new species in a day : #213
> 

I really like this!

alexbelgium, 2025-05-11:
I think we can close this one as the feature was merged

wenestvedt, 2025-07-01:
Sorry, that's on me. Thank you for this!!

closed 2025-07-01 by wenestvedt