Nachtzuster/BirdNET-Pi #147: Label cut-off on Overview screen

dalogue1, 2024-07-13:
Is there a way to fix the label width so the name will not be cut off on the Overview screen?

See the label below for Common Nighthawk:

[image]

lloydbayley, 2024-07-16:
I've had a good look into this and I can't replicate it with the birds I have here. I have seen this on other installations from other people but not here. It's part of the daily_plot.php which creates the image and the code is scary. @Nachtzuster is this something you are familiar with? daily_plot.php has always scared me!

Nachtzuster, 2024-07-16:
@lloydbayley fortunately it is _not_ there :smile: 
This line conditionally wraps the label>16. (It does not take in account letter width)
https://github.com/Nachtzuster/BirdNET-Pi/blob/09dac978aff45ee50cec86b753fbccaf3fcd3add/scripts/daily_plot.py#L100C66-L100C69

'Common Nighthawk' should really be wrapped, but it is a letter short to do so.
We can't just wrap sooner or we'll get more labels like 'Northern Cardinal': it could fit one line though it is even one letter _longer_ than 'Common Nighthawk'

I've added some estimation on wrap width

@dalogue1 could you test this branch

```
git remote remove origin
git remote add origin https://github.com/Nachtzuster/BirdNET-Pi.git
git fetch
./scripts/update_birdnet.sh -b text_wrap
```

dalogue1, 2024-07-16:
I am happy to help.

Are you saying that I should run those commands in a terminal window?

(Can I assume this will not affect any configuration or history?)

dalogue1, 2024-07-17:
Looks good!  Thank you.

[image]

dalogue1, 2024-07-17:
It is fixed

lloydbayley, 2024-07-17:
Sorry @Nachtzuster, it was late. I meant daily_plot.py not php. That's the line I was looking at to debug so I'm going to have a look at your code to learn something.

DougWelch, 2024-07-17:
> Caution: External email.

Folks,

I'm glad someone else is experiencing this. It is a very minor annoyance 
with a really wonderful program! I have attached an image showing the 
Downy Woodpecker label mostly cutoff at the "D". It does happen for 
Common Nighthawk and other labels too. Changing the textwrap number 
didn't cure it.

Thanks in advance.

Cheers,
Doug

--------------SR5slOwl77d9VDbCDfXhzoel
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <br>
    Folks,<br>
    <br>
    I'm glad someone else is experiencing this. It is a very minor
    annoyance with a really wonderful program! I have attached an image
    showing the Downy Woodpecker label mostly cutoff at the &quot;D&quot;. It does
    happen for Common Nighthawk and other labels too. Changing the
    textwrap number didn't cure it.<br>
    <br>
    Thanks in advance.<br>
    <br>
    Cheers,<br>
    Doug<br>
    <br>
    <div class="moz-cite-prefix">On 7/17/24 00:39, Lloyd Bayley wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      
      <table width="100%" cellspacing="0" cellpadding="0" border="0" align="left">
        <tbody>
          <tr>
            <td style="background:#7A003C;padding:5pt 2pt 5pt 2pt"><br>
            </td>
            <td cellpadding="7px 6px 7px 15px" style="background:#ff000;padding:5pt 4pt 5pt 12pt;word-wrap:break-word" width="100%">
              <div style="color:#000000;"><span style="color:#000000; font-weight:bold;">Caution:
                </span>External email. </div>
            </td>
          </tr>
        </tbody>
      </table>
      <br>
      <div>
        <p dir="auto">Sorry <a class="user-mention notranslate" data-hovercard-type="user" data-hovercard-url="/users/Nachtzuster/hovercard" data-octo-click="hovercard-link-click" data-octo-dimensions="link_type:self" href="https://github.com/Nachtzuster" moz-do-not-send="true">
            @Nachtzuster</a>, it was late. I meant daily_plot.py not
          php. That's the line I was looking at to debug so I'm going to
          have a look at your code to learn something.</p>
        <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
            unsubscribe</a>.<br>
          thread.[image]<span style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
            ID: <span>&lt;Nachtzuster/BirdNET-Pi/issues/147/2232399924</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
        <script type="application/ld+json">[
{
"target": "https://github.com/Nachtzuster/BirdNET-Pi/issues/147#issuecomment-2232399924",
"url": "https://github.com/Nachtzuster/BirdNET-Pi/issues/147#issuecomment-2232399924",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
[... 712 lines cut ...]
AecAt2uHLgq7iMQjmwXbQrI4Yl0oaT66/YEF297MnaoZ1KeJ8IsoQz8PkQAJVF4CuAL0/WbE4R93
/jV5BWak4BFcmwmdC68JpoZ4BHdy2cEreyJxHRAma8roJqZX9v9+FB7iRW09i7yR9lO6CXluMqpc
2hmBj46eDlLmun8+SCTfZVa/gwu2//HOUkdf5+Cp3XFvgEslYhZQBRnM0VQosi7DAB4iARKQnwD9
b/lZUVJNCSDqk/Iiycy+MEqNHEenmi4iQ/EbkxqfLGH3qN/fF600gEfDwlHplM2y6i1WNsAiX0gZ
v3n8StT2syNWvCcsRFBW+qmHBEhAxQQmbp0h0WK32QNEJeaOlv2/GyVtUu2OQfhIlOP6gI+0MEpc
gzxHrZkifWhG6HzpwqG/jJMuFEqwfMrbqyeLjtbqUFe0jWgCg9/FcWM5CZQrAeaflCteKlcFAUyd
xGq7Pq394W0j2QOTooQPliDQ+de9LrTExq1wjhFWABAELBytEFgqLHe3f3YjTrAVeZk7Pt2Ax7jC
ruQUKKwb4G7/OPqhqGPYlhE9Qq45ngVjqhYWDh+xYpK5oxXmP6kCCtsgARIggeIJIB0F816wBHhg
twbFS/EICZBAeRFg/Lu8yFJvuRLAQiVwkZHgcfNkNNKve305BEuONBjUAq95g9sNf/rKvvMxEXfw
SgtDLLalpRV74Z6uvi4cZfdG3vvmbWn/fnddQz0sGWbtYoMZlg0GNj/47XYkfyOlO3LLmafXHxlb
m8J+I0sTTF1C4iYmZYq6g5lM68YtC19/0rOpL16ccffMTczaLK6zeBZ86tdDqS9TArrWT4yNT372
CtqKE2Y5CZAACaiGwMllB5Ch3vGjXsJEF9U0ylZIgAREBOh/82SolASwXAk+SK+sVqv6oB/HVK/r
gW40HtIyOy3z6KLdmamZyELpOWcwcjGRio15SyeW7n9663G3//Xv/tnAIz/u3vrxWjjleIGcsOpt
zeA6KfHJJ389iPcwwwXvt2Ck8Kqd5qPbwUd/cjV2yJLCRVSEP+RQ4k1yWOjw9Kojls422EZJcRCN
LEzwqrmQ3w5hrRWs89VgYAu/tnzPRXG0WE4CJKAiAgMWjlZRS2yGBEigKAJaRWaSFSXJMhIgARIg
ARIgARIgARIgAWUJMP9bWYKsTwIkQAIkQAIkQAIkQALyE6D/LT8rSpIACZAACZAACZAACZCAsgTo
fytLkPVJgARIgARIgARIgARIQH4C9L/lZ0VJEiABEiABEiABEiABElCWAP1vZQmyPgmQAAmQAAmQ
AAmQAAnIT4D+t/ysKEkCJEACJEACJEACJEACyhKg/60sQdYnARIgARIgARIgARIgAfkJ0P+WnxUl
SYAESIAESIAESIAESEBZAvS/lSXI+iRAAiRAAiRAAiRAAiQgPwH63/KzoiQJkAAJkAAJkAAJkAAJ
KEuA/reyBFmfBEiABEiABEiABEiABOQnQP9bflaUJAESIAESIAESIAESIAFlCdD/VpYg65MACZAA
CZAACZAACZCA/ATof8vPipIkQAIkQAIkQAIkQAIkoCwB+t/KEmR9EiABEiABEiABEiABEpCfAP1v
+VlRkgRIgARIgARIgARIgASUJUD/W1mCrE8CJEACJEACJEACJEAC8hP4P4LUA1bW/nboAAAAAElF
TkSuQmCC

--------------6K4LMM05udV7d0HIhcID6wkV--

Nachtzuster, 2024-07-17:
@DougWelch just checked Downy Woodpecker: that would be wrapped too with the change. Any other species?

Nachtzuster, 2024-07-17:
> Kind of a big club for a small but annoying problem.

quite... Not going there, I think

DougWelch, 2024-07-17:
Hi,

|git remote remove origin git remote add origin 
https://github.com/Nachtzuster/BirdNET-Pi.git git fetch 
./scripts/update_birdnet.sh -b text_wrap |

|cured the issue for me!!! Thanks!!!
|||
The only other species I had seen this happen for was Common Nighthawk.

Cheers,
Doug

--------------gFGGFyKkWnw4eORTr3Q0lMeF
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <br>
    Hi,<br>
    <pre class="notranslate"><code class="notranslate">git remote remove origin
git remote add origin <a class="moz-txt-link-freetext" href="https://github.com/Nachtzuster/BirdNET-Pi.git">https://github.com/Nachtzuster/BirdNET-Pi.git</a>
git fetch
./scripts/update_birdnet.sh -b text_wrap
</code></pre>
    <code>cured the issue for me!!! Thanks!!!<br>
    </code><code class="notranslate"></code><br>
    The only other species I had seen this happen for was Common
    Nighthawk.<br>
    <br>
    Cheers,<br>
    Doug<br>
    <br>
    <br>
    <div class="moz-cite-prefix">On 7/17/24 11:49, Nachtzuster wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      
      <table width="100%" cellspacing="0" cellpadding="0" border="0" align="left">
        <tbody>
          <tr>
            <td style="background:#7A003C;padding:5pt 2pt 5pt 2pt"><br>
            </td>
            <td cellpadding="7px 6px 7px 15px" style="background:#ff000;padding:5pt 4pt 5pt 12pt;word-wrap:break-word" width="100%">
              <div style="color:#000000;"><span style="color:#000000; font-weight:bold;">Caution:
                </span>External email. </div>
            </td>
          </tr>
        </tbody>
      </table>
      <br>
      <div>
        <p dir="auto"><a class="user-mention notranslate" data-hovercard-type="user" data-hovercard-url="/users/DougWelch/hovercard" data-octo-click="hovercard-link-click" data-octo-dimensions="link_type:self" href="https://github.com/DougWelch" ***@***.***</a>
          just checked Downy Woodpecker: that would be wrapped too with
          the change. Any other species?</p>
        <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
            unsubscribe</a>.<br>
            ID: <span>&lt;Nachtzuster/BirdNET-Pi/issues/147/2233643783</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
        <script type="application/ld+json">[
{
"target": "https://github.com/Nachtzuster/BirdNET-Pi/issues/147#issuecomment-2233643783",
"url": "https://github.com/Nachtzuster/BirdNET-Pi/issues/147#issuecomment-2233643783",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script></div>
    </blockquote>
    <br>
  </body>
</html>

--------------gFGGFyKkWnw4eORTr3Q0lMeF--

Nachtzuster, 2024-07-17:
Seems the simple guestimation works well enough, the fix is merged.
I've added j to the 'thin' list.
I've left Q out, with this Roboto font, the 'tail' most likely ends up under the next letter making take it no more room than O

@dalogue1, @DougWelch thanks for testing! Tools->System Controls->Update, and you'll be back on main

dalogue1, 2024-07-17:
Happy to help.

closed 2024-07-18 by Nachtzuster