Nachtzuster/BirdNET-Pi #148: Fresh install needs pipewire for audio

Hiemstra87, 2024-07-14:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
After a fresh install audio is not getting through to BirdNET. PulseAudio is not correctly passing the audio through. 

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install on a PI 4B, using bookworm-lite
2. Curl from nachtzuster repository
3. Checked to see the correct device through a record -l

**Expected behavior**
Sound coming on for processing. Spectogram viewer / live audio feed passes live sound through.

**Screenshots**
N/A

**Additional context**

**Your build**
Using a Ugreen USB soundcard:
https://nl.ugreen.com/collections/audio-videokabel/products/80864

After installing pipewire with "sudo apt install pipewire-pulse pipewire-jack" audio was coming in again.

Might be just my install, but if this is reproducable by others we might have to look at PulseAudio.

</details>

lloydbayley, 2024-07-16:
Just did an identical install from scratch on a test Pi and it works first time. Sounds like it's either something to do with your config, or something peculiar to the ugreen sound card. (I don't have one to test).

Hiemstra87, 2024-07-16:
Thanks for testing, should be fine then for other users.

closed 2024-07-16 by Hiemstra87

alexbelgium, 2024-07-25:
It all depends on the system on which you install birdnet-pi. It is configured for pulseaudio but perhaps not all systems have it. Pipewire is a more modern pulseaudio alternative