Nachtzuster/BirdNET-Pi #157: Duplicate telegram notificaitons

alexbelgium, 2024-07-25:
1. Activate telegram notification
2. Wait for notification
3. Two notifications will be received : one with text, one with picture
 
**Expected behavior**
Single notification with bird detection and picture.
It seems to me the code is made that way

**Your build**
Birdnet-pi aarch64 through HA addon

Nachtzuster, 2024-08-03:
For email this does not occur: just one message, I think this might just be how be Apprise handles 'attachments' for telegram?

alexbelgium, 2024-08-03:
Perhaps... Well then if it is specific to telegram and it is a normal apprise behavior I'll close it as correcting this would be too complex for the benefit. Thanks for your message

closed 2024-08-03 by alexbelgium

Nachtzuster, 2024-08-03:
Perhaps you could ask over https://github.com/caronc/apprise it might still be a bug.

alexbelgium, 2024-08-03:
Actually you were right, their [telegram wiki](https://github.com/caronc/apprise/wiki/Notify_telegram) from the repo you linked states : `Due to the services limitations, Telegram doesn't allow you to post an image inline with a text message. But you can send a message that just contains an image. Since receiving 2 messages for every 1 notice could be annoying to some, this has been made an option that defaults to being disabled.`

Thanks!