Nachtzuster/BirdNET-Pi #161: BirdWeather Reporting Threshold Feature

dalogue1, 2024-07-31:
I would like to suggest that a detection of a bird should not be reported to BirdWeather unless a specified threshold for the past 24 hours has been detected.  (A default of 0 would behave exactly as it does now.  In fact, the code would not even perform a query to the database if the threshold is 0.)

For instance, if I have a detection of a Bald Eagle and the threshold is set to 1 then it will not report the detection to BirdWeather unless it is the 2nd or more for the day.  This avoids the many reporting of detections of one-off species that are just normal.  I do not mind them in my local database but would prefer not to report them to BirdWeather publicly. (Vanity of Vanities)  

This is a simple change to the reporting.py script as well as a way to enter the threshold in the settings interface.

alexbelgium, 2024-08-01:
I could add in this PR https://github.com/Nachtzuster/BirdNET-Pi/pull/158 a flag so only confirmed species (meaning that you have at least once confirmed that this species really exists) are uploaded to birdweather.

dalogue1, 2024-08-01:
> I could add in this PR #158 a flag so only confirmed species (meaning that you have at least once confirmed that this species really exists) are uploaded to birdweather but birdweather by design looks more at volume than exactitude of observations

That is a really good idea to use the confirmed species flag instead of a daily threshold!  This would prevent uploaded a Bald Eagle detection by mistake.

alexbelgium, 2024-08-07:
Indeed. I see the "confirmed species" feature as a key to other elements, but I propose to wait first for it to be merged as a standalone feature before adding additional elements on top to avoid complexification.

dalogue1, 2024-08-26:
How do you set the Enhancement flag?

lloydbayley, 2024-08-27:
Ahh...Sometimes, it used to let the OP set the flag but perhaps it's only the repository-owner that can set it now. Doesn't matter. :)

closed 2025-09-29 by dalogue1