Nachtzuster/BirdNET-Pi #165: NEW FEATURE REQUEST: Notification Link to Wikipedia on Bird (or other site)

flewid, 2024-08-04:
On the notification settings, it would be cool to have a {link_to_wikipedia} option so when we get a notification of a species, we can click to somewhere to read about the species etc.

Emmo213, 2024-08-06:
PR opened

UlrichThiess, 2024-08-20:
In Home Assistant i create a link with the $sciname

draft: https://de.wikipedia.org/wiki/($sciname).replace(' ', '_')

Any automatism should recognize your language.

Emmo213, 2024-08-22:
I like your idea about using the language specific version of Wiki but that may be an enhancement in and of itself since the other places where Wiki is referenced (overview, play, stats, todays detections) don't do that currently.

Nachtzuster, 2024-08-25:
Maybe having a url-encoded version of $sci_name and $com_name, like $sci_name_url and $com_name_url would offer a more generic solution, then one could put this in the notification: `https://de.wikipedia.org/w/index.php?search=$sci_name_url`
If someone want a link to some other info site that would be possible too.

Emmo213, 2024-08-25:
Potentially except Wiki, allaboutbirds, and eBird all have different formats for linking to a bird. 

$sci_name_url would only work with Wiki, $com_name_url would only work with allaboutbirds, and neither would work with eBird. 

We could create a $infourl which would figure out the URL to use based on your information settings although that wouldn't respond with Wiki, which was the actual request, unless you wanted to add Wiki as an information option, but now we're getting into scope creep.