Nachtzuster/BirdNET-Pi #173: FEATURE REQUEST: rsync/scp site to remote server

flewid, 2024-08-15:
I run a variety of other similar things (AllSky, WeeWX Weather) and I'm able to rsync/scp/ftp the website to an external server. It would be really nice if we could do this with Birdnet-PI instead of having to port forward on our internal network. Thought this would be a a neat addition to this project. Happy to help with it if need be.

tomtom215, 2024-09-16:
Not sure if many people would benefit from such a feature, but an alternative approach you could take is replicating the sqlite DB to another installation and streaming the DB updates via SFTP with LiteStream https://github.com/benbjohnson/litestream

flewid, 2024-09-17:
Oh this is interesting! Thank you. My purpose was to not have to port forward or have dyn-dns setup so people could access the site, as well as I was thinking that the 'synced' site would not have any admin interface, solely the front-end so friends/family could view the birds captured at my location. Though this would require 'two sets of files' for the different site types so more complex. This is how weewx does it for the weather info I push and thought it'd be cool for this as well. I'll take a look at LiteStream for now though, since the site doesn't change very often I can probably create my own copy of it without the admin side, and just sync the database.

Nachtzuster, 2025-01-18:
> My purpose was to not have to port forward or have dyn-dns setup so people could access the site, as well as I was thinking that the 'synced' site would not have any admin interface, solely the front-end so friends/family could view the birds captured at my location. 

I imagine having a sufficiently complex `BirdNET-Pi Password` would get you most of what you want, unauthorised users not able to mess with the system.
Tools->Settings->Advanced