Nachtzuster/BirdNET-Pi #177: Feature Request: Name RTSP Streams

UlrichThiess, 2024-08-20:
I have many mics via Raspberry PI RTSP connected to my BirdNET-Pi. But i can't identify where the bird was.

I planned to add some mics into my hunting ground to care for the birds there. In Germany the capercaillie is protected and needs special care.

Nachtzuster, 2024-08-25:
The sound and spectrum file name should have the RTSP identifier in there ,I believe: Gray_Wagtail-84-2024-04-22-birdnet-**RTSP_1**-09:29:05.mp3, so in this case the first RTSP steam
Is that what you're looking for?

UlrichThiess, 2024-08-25:
Yeah! Thank a lot. Thats all what i want.

Danke.

closed 2024-08-25 by UlrichThiess

UlrichThiess, 2024-08-26:
@Nachtzuster Btw. a name would be nicer in case the order is changed or the name is further processed.