Nachtzuster/BirdNET-Pi #196: Wrong latin name on Common Starling recordings

Hiemstra87, 2024-09-20:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
All my Common Starling (Sturnus vulgaris) recordings get the scientific name for Daurian Starling (Agropsar sturninus). (Dutch name is correct though)

I don't know if this is something that can be fixed in the BirdNET application or if this is wrong in the TFlite.

**Screenshots**
If applicable, add screenshots to help explain your problem.

</details>

alexbelgium, 2024-09-20:
Have you updated to latest? It would have supposed to be solved by https://github.com/Nachtzuster/BirdNET-Pi/pull/187

Hiemstra87, 2024-09-20:
Ah, silly me. Haven't noticed that issue. Fixed with the update indeed. Thanks for the quick response.

closed 2024-09-20 by Hiemstra87