Nachtzuster/BirdNET-Pi #200: Tools -> Settings -> Advanced Settings gives PHP error

gerdesj, 2024-09-28:
New install on a RPi 4 minimal RaspiOS. 

Tools -> Settings -> Advanced settings, results in this error:
`Fatal error: Uncaught TypeError: count(): Argument #1 ($value) must be of type Countable|array, bool given in /home/local-admin/BirdNET-Pi/scripts/advanced.php:249 Stack trace: #0 /home/local-admin/BirdNET-Pi/homepage/views.php(151): include() #1 {main} thrown in /home/local-admin/BirdNET-Pi/scripts/advanced.php on line 249
`

Creating an empty file at ~/BirdNET-Pi/model/labels.txt makes the error go away because the function returns 0 instead of failing and returning (presumably) false.

lloydbayley, 2024-09-28:
That's really weird. The ~/BirdNET-Pi/model/labels.txt is created during the install and always has been. It should have been there.
Did you have any problems with the install?
Was it a fresh install from this repo or a crossgrade from another one?

gerdesj, 2024-09-28:
Fresh install from here.  I've blown it away using the uninstall routine and done a new one with the same result.  I will do some investigation, I've been a Linux sysadmin for 25 years so really ought to be able to work out what is going on!

A quick run through the install log shows no errors.

gerdesj, 2024-09-28:
... found install__language_label.sh_ and opened a bottle of claret ...

gerdesj, 2024-09-28:
The line 246 in _advanced.php_ is:

`$count_labels = count(file($home."/BirdNET-Pi/model/labels.txt"));`

In the installation log:

`+ sudo -u local-admin ln -fs /home/local-admin/BirdNET-Pi/model/labels.txt /home/local-admin/BirdNET-Pi/scripts`

That is the only reference in there to labels.txt.  That file is created by install_language_label.sh ...

Oh I've had enough of using the Github web interface.  Its pretty, but shit for this sort of thing.  I'll check out locally ...

lloydbayley, 2024-09-29:
Claret certainly helps lubricate the diagnostic thought processes!
Certainly is strange because that's the file that creates it and it's called via the newinstaller.sh which you ran initially.
Not too sure what would either stop that or remove the file.

Just out of interest, my ~/BirdNET-Pi/model directory looks like this:

[image]

Is it similar to yours? (Bearing it mind I've been running this thing for a looooooooong time and many iterations and a repo swap, so there could be some leftover detritus!) :)

gerdesj, 2024-09-29:
Mine looks like this.  I created tmp and labels.txt

```
birdnetpi:~/BirdNET-Pi/model$ ls -l
total 107004
-rw-rw-r-- 1 local-admin local-admin 56681380 Sep 28 21:27 BirdNET_6K_GLOBAL_MODEL.tflite
-rw-rw-r-- 1 local-admin local-admin  7071440 Sep 28 21:27 BirdNET_GLOBAL_6K_V2.4_MData_Model_FP16.tflite
-rw-rw-r-- 1 local-admin local-admin 14770468 Sep 28 21:27 BirdNET_GLOBAL_6K_V2.4_MData_Model_V2_FP16.tflite
-rw-rw-r-- 1 local-admin local-admin 25932528 Sep 28 21:27 BirdNET_GLOBAL_6K_V2.4_Model_FP16.tflite
-rw-rw-r-- 1 local-admin local-admin  2173578 Sep 28 21:27 labels_l18n.zip
-rw-rw-r-- 1 local-admin local-admin      649 Sep 28 21:27 labels_lang.txt
-rw-rw-r-- 1 local-admin local-admin  2661024 Sep 28 21:27 labels_nm.zip
-rw-r--r-- 1 local-admin local-admin   253728 Sep 28 22:09 labels.txt
drwxr-xr-x 2 local-admin local-admin     4096 Sep 28 22:08 tmp

```

lloydbayley, 2024-09-29:
Interesting. Looks pretty standard to me. 
Anyone else have any ideas?

gerdesj, 2024-09-29:
> Interesting. Looks pretty standard to me. Anyone else have any ideas?

I'm going to blat the Pi and start again this evening (GMT/UTC+1)

lloydbayley, 2024-09-29:
Copy that. Shout at it when you do it!

gerdesj, 2024-10-04:
OK so that worked - start from scratch.  

This time I changed my admin user to not have a - in its name and I did not mount /home on a network block device.  I'll also need to check out any code changes between now and then.

lloydbayley, 2024-10-07:
Oh,...Can't see why a - would make a difference....the /home, although unusual, I wouldn't have thought would be the issue. Perhaps a fresh install worked because on the odd occasion, an install relying on web-components may not completely complete properly because a server was unavailable/offline. Who knows. It's the internet! :) Glad it's working for you!

gerdesj, 2024-10-20:
It's unlikely to be an issue with BNP.

closed 2024-10-20 by gerdesj