Nachtzuster/BirdNET-Pi #201: PRs I think are ready to accept.

lloydbayley, 2024-09-28:
I've been testing out the following PR's and they seem to be stable.
Any other opinions?

198
195
194
190
166

Just trying to triage things a little as they seem to be piling up.

If all in agreement, perhaps @Nachtzuster might look at merging them?

Emmo213, 2024-10-01:
I know this was just closed but for PRs which have been open for awhile it'd be nice to have feedback on if something is wrong with them or if they're just not wanted. Leaving them open with no feedback isn't helpful to the requestor.

lloydbayley, 2024-10-01:
I only closed it because things had been done with no indication and my issue was adding to the clutter. I guess I've upset the apple cart by asking for action. Again.

Emmo213, 2024-10-01:
My comment wasn't geared towards you, but just in general. Like I have a few PRs opened but aren't merged and I don't know why. Is there something wrong with the code (and not just "I would have done it differently") or are the changes unwanted? The one, the wiki link, I didn't need as an enhancement but somebody requested it and I thought "that's an easy enhancement I can make for somebody" but it hasn't been merged. I can close it but then we should tell the person that requested it that it's not going to happen.

lloydbayley, 2024-10-01:
Nono, I realise that it wasn't geared toward me. I, too, was hoping for some feedback but alas. I'll open this again to see.

lloydbayley, 2024-10-01:
@alexbelgium, Have you any opinions on those PR's?

alexbelgium, 2024-10-01:
> My comment wasn't geared towards you, but just in general. Like I have a few PRs opened but aren't merged and I don't know why. Is there something wrong with the code (and not just "I would have [...]

I do remember Nachtzuster saying somewhere that he is quite busy and has little time for the moment so only merges things that don't need in depth review.

The key element is that, as for any github dev, it is all done on top of busy lives and I understand if he has other priorities in his life to look at for the moment. Especially as I saw from his past interactions that he puts a point to ensure code stays lean & efficient - so actually I prefer for him indeed to take the time that he needs and hopefully merge it at some point when he'll have time.

Let's not forget that anyway we can merge locally on our systems PRs so we can still benefit personnaly even if it is not merged in the main repo !

> @alexbelgium, Have you any opinions on those PR's? Would be interested in your thoughts even though they have been merged.

Not yet merged : 
- https://github.com/Nachtzuster/BirdNET-Pi/pull/166/files : I don't use this but code looks good!

Merged : 
- https://github.com/Nachtzuster/BirdNET-Pi/pull/194/files : good to protect other *.db files, not needed for base functions but can help for multiple databases
- https://github.com/Nachtzuster/BirdNET-Pi/pull/190/files : useful to have more log, closes https://github.com/Nachtzuster/BirdNET-Pi/issues/138
- https://github.com/Nachtzuster/BirdNET-Pi/pull/195/files : always good for readme updates
- https://github.com/Nachtzuster/BirdNET-Pi/pull/198/files : useful feature!

alexbelgium, 2024-10-01:
> @alexbelgium, Have you any opinions on those PR's? Would be interested in your thoughts even though they have been merged.

Not yet merged : 
- https://github.com/Nachtzuster/BirdNET-Pi/pull/166/files : I don't use this but code looks good!

Merged : 
- https://github.com/Nachtzuster/BirdNET-Pi/pull/194/files : good to protect other *.db files, not needed for base functions but can help for multiple databases
- https://github.com/Nachtzuster/BirdNET-Pi/pull/190/files : useful to have more log, closes https://github.com/Nachtzuster/BirdNET-Pi/issues/138
- https://github.com/Nachtzuster/BirdNET-Pi/pull/195/files : always good for readme updates
- https://github.com/Nachtzuster/BirdNET-Pi/pull/198/files : useful feature!

lloydbayley, 2024-10-01:
Right.  I'll stay out of it.
Just wanted to make sure that the repo wasn't dying as there is no feedback when there is action.

alexbelgium, 2024-10-01:
Well I didn't really want to push back, but as a repo maintainer myself I emphasize with the complexity of managing expectations while also having lots of things in parallel

But I'm sure your support is appreciated, for example your post here led to the merging of most of the PR linked

lloydbayley, 2024-10-04:
Closed as it's pointless leaving it open because there isn't likely to be anything constructive out of it by the looks of things. I've unsubscribed from the thread. I'm sorry I pushed the issue in the first place. Not apologetic, just sorry I started it to cause the drama...

closed 2024-10-04 by lloydbayley