Nachtzuster/BirdNET-Pi #202: 502 Bad Gateway

tsvet551, 2024-09-28:
Hi! 
I installed the program, updated it, and the web interface periodically disappears: 502 Bad Gateway. You have to restart the program every day or several times a day. What could it be, colleagues? How to fix it?

lloydbayley, 2024-09-29:
Quick question. Are you using an Ngix server for redirection or is it entirely native?

tsvet551, 2024-09-29:
Greetings! Thanks for the answer!
Unfortunately, I am not a great specialist. In short: I installed the SD-card on the BirdNET-Pi command, the address was assigned to the Keenetic 4G IP router network, at first the page opens by IP, but then I can't get to the page- it gives an error 503, it continues to work on the network. I did not install any other software.
Ngix server has not been used anywhere, only standard tools.
Sincerely, Alexander!

lloydbayley, 2024-09-29:
Ok. In the settings for BirdNet, under the Advanced settings, check the Custom URL field.
Ideally, this should be set to the address you want to use for it.
However, the restarting to fix the problem has me a bit stumped...

tsvet551, 2024-09-29:
Thanks for the advice, I'll try! 
Yes, it is a restart using a power outage, so I can't see and log in to the BirdNET-Pi web page.

tsvet551, 2024-10-03:
> Ok. In the settings for BirdNet, under the Advanced settings, check the Custom URL field. Ideally, this should be set to the address you want to use for it. However, the restarting to fix the problem has me a bit stumped...

Greetings, colleague!
After your prompt, the error began to appear less frequently, but it still appears. It got better, thanks!

lloydbayley, 2024-10-04:
Well, I suppose it's good news but not entirely a complete fix. I am starting to think it could be a networking issue on your end that may have resolved itself. Glad it's working for you!

tsvet551, 2024-10-24:
Greetings!
A complete solution to the problem. The problems were related to the Wi-Fi. I turned off the Wi-Fi, connected the LAN cable - the problem disappeared, it has been working without failures for 3 weeks. Your advice on setting up a BirdNET also helped. Later I saw the topic [https://github.com/Nachtzuster/BirdNET-Pi/discussions/218](url) apparently my problem is related to the same thing. Thank you very much! Sincerely, Alexander!

closed 2024-11-04 by Nachtzuster