Nachtzuster/BirdNET-Pi #227: Store actual detection time in DB instead of audio file start time

tvoirand, 2024-11-04:
Hi,

As of now, when storing detections in the DB, the detection time is set to the start time of the original audio file (which is the file present in “/home/$USER/BirdSongs/StreamData” by default, before it's "extracted").

Therefore, several detections generated from the same original audio file all share the same detection time, even though these detections actually occurred at different moments within that file. Additionally, the timestamps included in the filenames of the “extracted” audio files are also identical, even though each file corresponds to a unique interval within the original full-length file.

To address this, the detection time stored in the DB could be adjusted to reflect the actual detection time, by adding the delay in seconds between the start of the audio file and the start of the detection.

This involves changes in the `server.py`, `reporting.py` and `helper.py` scripts. I have tested this over the past few days on my station, and everything has been running smoothly without impacting any downstream logic of BirdNET-Pi. The modifications are visible [in this draft](https://github.com/tvoirand/BirdNET-Pi/tree/dates) if you want to have a look.

If maintainers of this project agree, I would be happy to submit a PR :).

Thanks!

Nachtzuster, 2024-11-09:
If it doesn't break things downstream, this would be a nice refinement :+1:

alexbelgium, 2024-11-12:
Looks nice! Not sure I fully understand how it works but I've merged it on my test fork and it seems to work fine. I had also noticed that some identifications were named identically despite being different. A PR on this repo could be nice indeed, thanks!

tvoirand, 2024-11-14:
Great! Thanks for your messages. PR submitted.

closed 2024-11-20 by Nachtzuster