Nachtzuster/BirdNET-Pi #243: Cannot Access Tools menu

stevedee, 2024-11-27:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
When I try to access Tools menu, it will not allow access using: **_birdnet_**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Menu selection
2. Click on Tools
3. User/password dialog, enter: birdnet
4. Won't let me in

**Additional context**
My problem is that the test for PHP AUTH USER & PW fails in: common.php

function is_authenticated() {
  //$ret = false;
  $ret = true; 
  if (isset($_SERVER['PHP_AUTH_USER']) && isset($_SERVER['PHP_AUTH_PW'])) {
    $config = get_config();
    $ret = ($_SERVER['PHP_AUTH_PW'] == $config['CADDY_PWD'] && $_SERVER['PHP_AUTH_USER'] == 'birdnet');
  }
  return $ret;
}

...so I have set **$ret = true** to defeat this code and allow me into Tools

**Your build**
Bookworm on a Pi 4 + 2GB RAM

Note: my testing so far seems to indicate that the system is otherwise working well, (..._and I don't really need user protection on the Tools menu_).

ashmckenzie, 2024-11-28:
@stevedee if you try the fix at https://github.com/Nachtzuster/BirdNET-Pi/pull/244, does this fix things for you?

rabede, 2024-11-28:
I had the same problem since a few days (there was some php8 update) and ashmckenzie's fix helped! Thanks!

stevedee, 2024-11-28:
Thanks @ashmckenzie, yes the user/pwd dialog _now_ works as it does on the McGuire system, but I've just noticed something else.
If you have 2 unsuccessful attempts to access Tools (e.g. by simply cancelling the dialog) you are then locked out until the remote browser is closed, then re-opened. Its the same on both the Nachzuster & McGuire systems, so possibly just the way PHP security works.

But I'm just going to stick with my mod as I'm the only one using this system, so don't need this security feature.

Thanks once again.

wilkart, 2024-11-29:
It looks like the empty password is not allowed anymore. I fixed the problem by setting the

BirdNET-Pi/birdnet.conf

```
## CADDY_PWD is the plaintext password (that will be hashed) and used to access
CADDY_PWD=mypassword
```

and running script
```
bash BirdNET-Pi/scripts/update_caddyfile.sh
```

closed 2024-12-01 by Nachtzuster