Nachtzuster/BirdNET-Pi #247: Import detections.sql (more than 2MB)

cesa030, 2024-11-30:
Hello,

I have had BirdnetPI running successfully for several months. Now I have to reinstall it on an RPi3. I didn't back up the birds.db back then. Instead, export the database to a detections.sql. The file is over 30 MB in size. When I want to import it again via "Tools - Database Maintenance" I get the error message that the file is too large (over 2 MB). 
"Too big POST data. Reduce the data or increase the 'post_max_size' configuration directive."

How can I adjust the value? or how can I import my detections.sql?

Thanks!
Best regards

cesa030, 2024-11-30:
I was able to import a *.sql-file via adminer webserver file. But now I get the following error when I click on species stats:

[image]

alexbelgium, 2024-12-07:
Looks like you have non-UTF8 characters in your database... If you don't have a lot you might perhaps modify the offending species by hand in Adminer with Tools > Database maintenance ; paste the "abnormal name" that is written here with the ? ; and modify in a batch mode all those names to the one with the correct umlaut ?

An alternative would be to search & replace all non UTF8 characters in the database using a mysql command in adminer which could be more thorough but a bit least easy (at least for me ;-) )?

closed 2024-12-09 by cesa030