Nachtzuster/BirdNET-Pi #251: BirdNET Analysis (stalled - backlog of 6697 files in ~/BirdSongs/StreamData/)

jhackeur, 2024-12-09:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
A clear and concise description of what the bug is.

BirdNet-PI is freshly installed on Debian 12, and never work for now.

**To Reproduce**
Steps to reproduce the behavior:
See logs in tools, services:
BirdNET Analysis (stalled - backlog of 6697 files in ~/BirdSongs/StreamData/)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Your build**
Debian 12 freshly installed with BirdNet-PI, using RSTP flow as audio source.
RSTP .wav files are coming inside folder /BirdSongs/StreamData/, but no analysis

Don't hesitate to ask me for more logs/infos if needed, and thanks in advance for your help !

alexbelgium, 2024-12-09:
Hi, and your log doesn't give an error message showing why the Birdnet-analysis service is blocked ? For example if you try to restart it ? It's just a python script

jhackeur, 2024-12-09:
@alexbelgium : arg... permission issue...again ---
[image]

jhackeur, 2024-12-09:
Hi Alex

Better, but not the only issue ---
[image]

Someone else here are using real linux and no raspi os ?
Because i just perform a fresh install of my debian 12, and apprently we need to do adapt somes part (like adding passworless) to be able to run this BirdNet-PI on stand system

alexbelgium, 2024-12-09:
Sorry, I'm the one confusing you. I'm using a real debian (based on linuxserver image) but I've hacked it a lot to make it work with docker so I'm not used to work on "clean systems" ;-)

In conclusion the only command that you should type is `/home/pi/BirdNET-Pi/birdnet/bin/python3 $HOME/BirdNET-Pi/scripts/birdnet_analysis.py` which is the command executed in the service file `$HOME/BirdNET-Pi/templates/birdnet_analysis.service`

I can replicate exactly what you see : 
```
root@db21ed7f-birdnet-pi:~/BirdNET-Pi/scripts# ./birdnet_analysis.py
bash: ./birdnet_analysis.py: Permission denied
root@db21ed7f-birdnet-pi:~/BirdNET-Pi/scripts# chmod 777 *.py && ./birdnet_analysis.py
./birdnet_analysis.py: line 1: import: command not found
./birdnet_analysis.py: line 2: import: command not found
./birdnet_analysis.py: line 3: import: command not found
./birdnet_analysis.py: line 4: import: command not found
./birdnet_analysis.py: line 5: import: command not found
./birdnet_analysis.py: line 6: import: command not found
./birdnet_analysis.py: line 7: import: command not found
./birdnet_analysis.py: line 8: from: command not found
./birdnet_analysis.py: line 9: from: command not found
./birdnet_analysis.py: line 11: import: command not found
./birdnet_analysis.py: line 12: from: command not found
./birdnet_analysis.py: line 14: from: command not found
./birdnet_analysis.py: line 15: from: command not found
./birdnet_analysis.py: line 16: from: command not found
ERROR:systemctl:Unknown operation =
./birdnet_analysis.py: line 21: syntax error near unexpected token `('
./birdnet_analysis.py: line 21: `log = logging.getLogger(__name__)'
root@db21ed7f-birdnet-pi:~/BirdNET-Pi/scripts# /home/pi/BirdNET-Pi/birdnet/bin/python3 $HOME/BirdNET-Pi/scripts/birdnet_analysis.py
[server][INFO] LOADING TF LITE MODEL...
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
[server][INFO] LOADING DONE!
^C[birdnet_analysis][INFO] Caught shutdown signal 2
[birdnet_analysis][INFO] backlog is 1
[birdnet_analysis][INFO] Analyzing /home/pi/BirdSongs/StreamData/2024-12-09-birdnet-RTSP_1-17:20:17.wav
```

jhackeur, 2024-12-09:
OK, so wrong syntax, make sens for this issue.

But the good syntax is not working for me:
[image]

Nachtzuster, 2024-12-09:
Hmmm, 'Illegal instruction', really? Ancient cpu? I know older AMD can throw that error. Or maybe a vm that does not pass the needed instructions?
You need a Haswell or newer cpu (AVX2 is the thing)

Also, the x86_64 stuff is really only there for developpers, for x86_64 it, can get rather hairy. fast.

Anyway, ou _might_ get away with replacing TFlite with the normal tensorflow from Pypi in the birdnet venv.
It'll be a bit slower and the old v1 model won't work, but that might not matter much.

If you're unsure what any of this means, you might be better off with an rPi :wink:

alexbelgium, 2024-12-09:
Are you using proxmox? There seems to be some additional steps indeed : https://github.com/Nachtzuster/BirdNET-Pi/discussions/55

Here is the full install protocol on proxmox using debian : https://github.com/Nachtzuster/BirdNET-Pi/discussions/111 and indeed they went through the same steps as you such as installing password less root.

But you perhaps only need this step to make it work on your system! 
```
pip3 uninstall tflite_runtime
pip3 install tensorflow
```

jhackeur, 2024-12-09:
@Nachtzuster traduction is not "illegal" but more "instruction denied"

But the info provided by @alexbelgium fix my issue !
FYI - we need to do the source activate/deactivate to get it working:
source BirdNET-Pi/birdnet/bin/activate
pip3 uninstall tflite_runtime
pip3 install tensorflow
deactivate

Parsing all the backlog from my VM, really cool !
Thanks a lot

closed 2024-12-09 by jhackeur

Nachtzuster, 2024-12-09:
@jhackeur so now I want to know: what does `lscpu` report?

alexbelgium, 2024-12-10:
> @jhackeur so now I want to know: what does `lscpu` report?

you're thinking of automatically installing the full tensorflow when specific settings or cpu are detected ? as this has now been reported by several persons on proxmox it would be interesting if this is feasible

jhackeur, 2024-12-10:
@Nachtzuster here are the info
[image]

Don't hesitate to ask me if you want more info :)

Nachtzuster, 2024-12-10:
@jhackeur thanks!
Looks like an old Sandy Bridge cpu so it doesn't have AVX2 support.. then 'Illegal instruction' makes total sense.

Nachtzuster, 2024-12-10:
@alexbelgium Not really, old pre-Haswell cpu's aren't supportable anyway. I've gathered what I know about x86_64 here https://github.com/Nachtzuster/BirdNET-Pi?tab=readme-ov-file#x86_64-support

alexbelgium, 2024-12-10:
> @alexbelgium Not really, old pre-Haswell cpu's aren't supportable anyway. I've gathered what I know about x86_64 here https://github.com/Nachtzuster/BirdNET-Pi?tab=readme-ov-file#x86_64-support

Couldn't we do something like that in the install script or do you think it's related to too limited cases ? :
```
# Check if CPU flags exist and do not contain avx2
if lscpu | grep -q "Flags" && ! lscpu | grep -q "avx2"; then
    echo "NON SUPPORTED CPU DETECTED"
    echo "Your CPU doesn't support avx2, the analyzer service may not work."
    echo "Trying to install tensorflow instead of tflite_runtime. If this doesn't work, your CPU is alas not supported."
    source $HOME/BirdNET-Pi/birdnet/bin/activate
    pip3 uninstall -y tflite_runtime
    pip3 install tensorflow
    deactivate
fi
```

alexbelgium, 2025-03-06:
I was wrong, actually I made it work on a non-avx2 system. This is now embedded in my addon & docker container of birdnet-pi

```
# Check if the CPU supports AVX2
if [[ "$(uname -m)" = "x86_64" ]]; then
  if lscpu | grep -q "Flags"; then
    if ! lscpu | grep -q "avx2"; then
        bashio::log.warning "NON SUPPORTED CPU DETECTED"
        bashio::log.warning "Your cpu doesn't support avx2, the analyzer service will likely won't work"
        bashio::log.warning "Trying to install tensorflow instead of tflite_runtime instead. This might take some time (up to 5 minutes)."
        bashio::log.warning "You could try also Birdnet-Go which should supports your cpu"
        source /home/pi/BirdNET-Pi/birdnet/bin/activate
        mkdir -p /home/pi/.cache/pip || true &>/dev/null
        chmod 777 /home/pi/.cache/pip || true &>/dev/null
        pip3 uninstall -y tflite_runtime
        pip install --upgrade packaging==23.2
        pip3 install --upgrade --force-reinstall "https://github.com/snowzach/tensorflow-multiarch/releases/download/v2.16.1/tensorflow-2.16.1-cp311-cp311-linux_x86_64.whl"
        deactivate
    fi
  fi
fi
```