Nachtzuster/BirdNET-Pi #259: Feature Request: use fixed/frozen 'Hours of Day' heading on detection table

stevedee, 2024-12-17:
My suggestion is to move the x-axis "Hours of Day" legend to the top of the detection chart/table in Overview, and make it fixed. So when you scroll down the table, the 'hours' remain in view.

alexbelgium, 2024-12-17:
Hi, as it is a picture it is not feasible with the current implementation. With the dynamic layout from https://github.com/Nachtzuster/BirdNET-Pi/pull/225 it could be possible though but with some change.

stevedee, 2024-12-17:
Hi Alex and thanks for that. I can see your problem.
The other approach would be to rework the page header. It may then be possible to fix 'hours' & 'detections' to the header section, and delete them from the bottom of the chart.