Nachtzuster/BirdNET-Pi #269: Live Stream does not work

DD4WH, 2025-01-04:
* installed the latest version on RPi 4 
* installation works well, but:
* Live Stream does not start, when I click on it
* instead, it shows 0:00 and stays there
* if I restart Live Stream Services in "Tools", I can listen to 20sec of audio and then audio stops. This is reproducible, but the audio is always the same 20 seconds
* Spectrum tab does  not work properly: I can see the specctrum window, but no audio is heard and no spectrogram data is visible. When I restart Live Stream Services, I can hear/see 20 seconds, but after that audio and spectrogram stops again

DD4WH, 2025-01-18:
thanks for considering my issue!

Before restart of livestream service:

[image]

After restart of livestream service:

[image]

Nachtzuster, 2025-01-18:
It seems the icecast service in not running maybe?
Could you post the output of `journalctl -eu icecast2.service` 
Also the install log might be useful, it is in the home directory - could you post that as well?

DD4WH, 2025-01-18:
[image]

not sure how to find the install log ? Could you give me a prompt for that?

Nachtzuster, 2025-01-18:
> not sure how to find the install log ? Could you give me a prompt for that?

this will spit out the entire log on your terminal
`cat ~/installation-*.txt`

DD4WH, 2025-01-18:
are you sure you want the entire log ;-) ???

DD4WH, 2025-01-18:
Maybe this is interesting enough? Let me know if you would like the entire log.

`+ config_icecast
+ '[' -f /etc/icecast2/icecast.xml ']'
+ cp /etc/icecast2/icecast.xml /etc/icecast2/icecast.xml.prebirdnetpi
+ sed -i 's/>admin</>birdnet</g' /etc/icecast2/icecast.xml
+ passwords=("source-" "relay-" "admin-" "master-" "")
+ for i in "${passwords[@]}"
+ sed -i 's/<source-password>.*<\/source-password>/<source-password>birdnetpi<\/                                                                                                             source-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<relay-password>.*<\/relay-password>/<relay-password>birdnetpi<\/rel                                                                                                             ay-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<admin-password>.*<\/admin-password>/<admin-password>birdnetpi<\/adm                                                                                                             in-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<master-password>.*<\/master-password>/<master-password>birdnetpi<\/                                                                                                             master-password>/g' /etc/icecast2/icecast.xml
+ for i in "${passwords[@]}"
+ sed -i 's/<password>.*<\/password>/<password>birdnetpi<\/password>/g' /etc/ice                                                                                                             cast2/icecast.xml
+ sed -i 's|

Nachtzuster, 2025-01-18:
Hmmm ... all this looks just fine.

Could you post the output of  `sudo netstat -tlpn` and `ps ux && date`

DD4WH, 2025-01-18:
Yes, it is a completely fresh install, that has been running for a few weeks now, but Live Stream and Spectrum view never worked properly.

```
birdie@birdnetpi:~/BirdNET-Pi$ sudo netstat -tlpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      977/gotty-aarch64
tcp        0      0 127.0.0.1:2019          0.0.0.0:*               LISTEN      774/caddy
tcp        0      0 127.0.0.1:8001          0.0.0.0:*               LISTEN      309443/icecast2
tcp        0      0 127.0.0.1:8000          0.0.0.0:*               LISTEN      309443/icecast2
tcp        0      0 127.0.0.1:8888          0.0.0.0:*               LISTEN      508/gotty-aarch64
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      630/sshd: /usr/sbin
tcp        0      0 127.0.0.1:8501          0.0.0.0:*               LISTEN      998/python3
tcp6       0      0 ::1:8501                :::*                    LISTEN      998/python3
tcp6       0      0 :::80                   :::*                    LISTEN      774/caddy
tcp6       0      0 :::22                   :::*                    LISTEN      630/sshd: /usr/sbin
```

```
birdie@birdnetpi:~/BirdNET-Pi$ ps ux && date
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
birdie       688  0.0  0.2  19596  9856 ?        Ss   Jan04   0:00 /lib/systemd/systemd --user
birdie       689  0.0  0.1 104740  4840 ?        S    Jan04   0:00 (sd-pam)
birdie       745  3.6  0.6 968584 26172 ?        S<sl Jan04 736:50 /usr/bin/pulseaudio --daemonize=no --log-target=journal
birdie       746  0.0  0.1   8636  4992 tty1     S+   Jan04   0:00 -bash
birdie       765  0.0  0.0   8236  3712 ?        Ss   Jan04   0:00 /usr/bin/dbus-daemon --session --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
birdie       934  1.6  6.9 721360 271296 ?       Ssl  Jan04 325:57 /home/birdie/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/daily_plot.py --daemon --sleep 2
birdie       937  0.0  0.0   6800  3072 ?        Ss   Jan04   0:00 bash /usr/local/bin/spectrogram.sh
birdie       940  0.0  0.0   2504  1280 ?        S    Jan04   0:09 inotifywait -m -e close_write /home/birdie/BirdSongs/StreamData/analyzing_now.txt
birdie       941  0.0  0.1  11920  6568 ?        S    Jan04   4:26 bash /usr/local/bin/spectrogram.sh
birdie       960  0.0  0.0   6800  3072 ?        Ss   Jan04   0:00 bash /usr/local/bin/birdnet_recording.sh
birdie       965  0.7  0.1 354236  6656 ?        Sl   Jan04 153:19 arecord -f S16_LE -c2 -r48000 -t wav --max-file-time 15 -D default --use-strftime /home/birdie/BirdSongs/StreamData/%F-bir
birdie       970 16.0  9.2 1652612 359644 ?      Rsl  Jan04 3235:40 /home/birdie/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/birdnet_analysis.py
birdie       974  0.0  0.0   6800  3072 ?        Ss   Jan04   0:00 bash /usr/local/bin/gotty --address localhost -p 8080 --path log --title-format BirdNET-Pi Log birdnet_log.sh
birdie       977  0.0  0.3 1377728 14592 ?       Sl   Jan04   1:35 gotty-aarch64 --address localhost -p 8080 --path log --title-format BirdNET-Pi Log birdnet_log.sh
birdie       998  0.0  5.7 988944 221332 ?       Ssl  Jan04   0:51 /home/birdie/BirdNET-Pi/birdnet/bin/python3 /home/birdie/BirdNET-Pi/birdnet/bin/streamlit run /home/birdie/BirdNET-Pi/scri
birdie    309449  0.0  0.0   6800  3072 ?        Ss   13:13   0:00 bash /usr/local/bin/livestream.sh
birdie    309450 16.0  1.2 609864 47360 ?        RLl  13:13  53:11 ffmpeg -nostdin -loglevel error -ac 2 -f alsa -i default -acodec libmp3lame -b:a 320k -ac 2 -content_type audio/mpeg -f mp
birdie    310423  0.0  0.1  19856  6560 ?        S    14:13   0:00 sshd: birdie@pts/0
birdie    310424  0.0  0.1   8756  5376 pts/0    Ss   14:13   0:00 -bash
birdie    314742  133  0.1  11136  4224 pts/0    R+   18:44   0:00 ps ux
Sat 18 Jan 18:44:32 CET 2025

```

Nachtzuster, 2025-01-19:
Sanity check: are you logged in?

When already logged in, you should be able to go to the Tools menu without being prompted for a password again.
Only when logged in the stream is available.

DD4WH, 2025-01-19:
Yes, I am logged in successfully. I have been using BirdNET-Pis in many different situations for many years now, so I am quite familiar with the usage.

DD4WH, 2025-01-19:
In my other installations, installed directly from Patrick's github some years ago, Live stream works. But I have never been able to get live stream running from your github.

Nachtzuster, 2025-01-19:
> Yes, I am logged in successfully. I have been using BirdNET-Pis in many different situations for many years now, so I am quite familiar with the usage.
> 
> Does live stream work for you in your installation?

Yes, I though I recognized the handle - as I said, sanity check, as I'm kinda stumped that it does not work for you

Nachtzuster, 2025-01-19:
> 
> Could you post the output of `sudo netstat -tlpn` and `ps ux && date`
> 

that is actually not very useful - could you try again with `sudo netstat -tlpn` and `ps aux | grep ice && date`

DD4WH, 2025-01-19:
Thank you for your continuous support! It would be really helpful to know how to use the Live Stream.

Here is the desired output:

```
birdie@birdnetpi:~/BirdNET-Pi$ sudo netstat -tlpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      977/gotty-aarch64
tcp        0      0 127.0.0.1:2019          0.0.0.0:*               LISTEN      774/caddy
tcp        0      0 127.0.0.1:8001          0.0.0.0:*               LISTEN      309443/icecast2
tcp        0      0 127.0.0.1:8000          0.0.0.0:*               LISTEN      309443/icecast2
tcp        0      0 127.0.0.1:8888          0.0.0.0:*               LISTEN      508/gotty-aarch64
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      630/sshd: /usr/sbin
tcp        0      0 127.0.0.1:8501          0.0.0.0:*               LISTEN      998/python3
tcp6       0      0 ::1:8501                :::*                    LISTEN      998/python3
tcp6       0      0 :::80                   :::*                    LISTEN      774/caddy
tcp6       0      0 :::22                   :::*                    LISTEN      630/sshd: /usr/sbin
birdie@birdnetpi:~/BirdNET-Pi$ ps aux | grep ice && date
nobody       475  0.0  0.0   5828  2560 ?        Ss   Jan04   0:12 /usr/sbin/thd --triggers /etc/triggerhappy/triggers.d/ --socket /run/thd.socket --user nobody --deviceglob /dev/input/event*
icecast2  309443  0.4  0.3 332424 12504 ?        Sl   Jan18   6:34 /usr/bin/icecast2 -b -c /etc/icecast2/icecast.xml
birdie    309450 15.9  1.2 609864 47488 ?        SLl  Jan18 252:59 ffmpeg -nostdin -loglevel error -ac 2 -f alsa -i default -acodec libmp3lame -b:a 320k -ac 2 -content_type audio/mpeg -f mp3 icecast://source:birdnetpi@localhost:8000/stream -re
birdie    334609  0.0  0.0   6088  1920 pts/0    S+   15:38   0:00 grep ice
Sun 19 Jan 15:38:32 CET 2025
```

Nachtzuster, 2025-01-25:
So both ffmpeg and icecast seem to be running fine.

What does `curl -v http://localhost/stream -u "birdnet:<YOUR PASSWORD>` return?

Then on a separate computer: what do you get in a private browser window when you hit
`http://birdnet:<YOUR PASSWORD>@<BIRDNET-PI IP>/stream`
Before hitting the address, turn on the developer tools to see if any interesting errors occur?

patmont, 2025-02-14:
I am also having this issue on a clean Rpi5 install with the same symptom. I can sometimes get audio after restarting the service.

To keep the ball rolling on the first part of your question:

`*   Trying 127.0.0.1:80...
* Connected to localhost (127.0.0.1) port 80 (#0)
* Server auth using Basic with user 'birdnet'
> GET /stream HTTP/1.1
> Host: localhost
> Authorization: Basic YmlyZG5ldDpiaXJkbmV0
> User-Agent: curl/7.88.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Access-Control-Allow-Origin: *
< Cache-Control: no-cache, no-store
< Content-Type: audio/mpeg
< Date: Fri, 14 Feb 2025 04:51:26 GMT
< Expires: Mon, 26 Jul 1997 05:00:00 GMT
< Icy-Metadata: 1
< Icy-Pub: 0
< Pragma: no-cache
< Server: Caddy
< Server: Icecast 2.4.4
< Transfer-Encoding: chunked
< 
Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
* Failure writing output to destination
* Failed reading the chunked-encoded stream
* Closing connection 0
`

zwk920, 2025-02-16:
I've been having this issue as well since switching from a USB mic to a Keyestudio ReSpeaker 2-mic HAT. The only way I've found to consistently fix this is to run the command "pkill -9 arecord" in the web terminal. Live Stream starts working again after a few seconds

DD4WH, 2025-02-16:
Sorry for my long silence, I was distracted by work obligations . . .
In the last weeks on my system Live Audio Stream was working sporadically, but I could not find any systematics . . .
What does `pkill -9 arecord` do exactly ?

zwk920, 2025-02-16:
When I had this issue, I found that `arecord` was using the mic and preventing BirdNet-Pi from using it. `pkill -9 arecord` kills all processes associated with `arecord`, and in my case, frees up the mic for BirdNet-Pi to use

DD4WH, 2025-02-16:
Thanks for the explanation! Is arecord used by the BirdNet-Pi? And, if not, what could cause arecord to run, if there is only BirdNET-Pi installed on the Pi? I use a headless installation on Pi4.

DD4WH, 2025-02-16:
hmm, the main BirdNET script: `birdnet_recording.sh` makes extensive use of `arecord`, so killing all processes would not be a very good idea? But maybe I am missing something here . . .

zwk920, 2025-02-16:
Yes, I realized that many of the BirdNet scripts use `arecord`, but this was after I had been consistently using `pkill -9 arecord` on start-up to fix the Live Recording issue. I'm not sure why running `pkill` works for me, I just know that it does work. I'm still trying to figure that out lol. For me, running `sudo reboot` restarts `arecord`, so you should be able to just restart your Pi if `pkill` doesnt have the intended effect