Nachtzuster/BirdNET-Pi #273: http://birdnetpi.local shows standard caddy page instead of birdnet-pi webui

hoover67, 2025-01-13:
Hi folks, 

first of all many thanks for picking up this excellent project and developing it further. 

I cloned the git repo on my new pi400 (setting this up for a friend), copied the newinstaller.sh script to ~/ and ran the newinstaller.sh script which apparently finished without errors. 

After the reboot, I only see the standard caddy "congratulations, it worked" page, not the birdnet-pi home page. (I have another pi400 running for 1,5 years without issues, so I'm not a complete noob when it comes to setting up birdnet-pi and / or birdnet go).

The caddy config file looks like this: 

```
http://  {
  root * /home/pi/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888
}

```

Release: 

```
# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

root@birdnetpi:~# uname -a 
Linux birdnetpi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

```

php-fpm is running fine according to systemctl, so I don't know what to try next in order to fix this problem. birdnet-go runs fine on the same host (I've bound it to port 8081), so the basic setup w/r to usb microphone etc should be ok. 

Thanks in advance for any help on this issue & keep up the great work! 

Uwe

hoover67, 2025-01-13:
PS: I failed to mention that recording, analysis etc. all work fine according to journalctl -f, it's apparently just the webUI that's acting up.

Nachtzuster, 2025-01-18:
Hi, does it work if you start the install as mentioned in the Readme?
`curl -s https://raw.githubusercontent.com/Nachtzuster/BirdNET-Pi/main/newinstaller.sh | bash` on a clean RPi?

hoover67, 2025-01-19:
Very weird, after trying a different (clean) install on a separate SD card (which broke because of a full / fs, the card only having 8GB :-P) the *original* card boots up fine in the same pi and the birdnet pi pages display as expected. Maybe all that was needed were a few reboots...

Cheers, Uwe

closed 2025-01-19 by hoover67