Nachtzuster/BirdNET-Pi #284: Notifications not being blocked for birds with apostrophes (') in their Common Names

zwk920, 2025-02-09:
Describe the bug
BirdNet-Pi sends discord notifications for birds with with apostrophes (') in their Common Name despite those birds being added to the notification exclude list (i.e. Anna's Hummingbird). Removing the apostrophe from the Common Name in both the labels.text file in /BirdNET-Pi/model and the notification exclude list in the Basic Settings page fixes this issue

To Reproduce
Steps to reproduce the behavior:

1. Add your Discord webhook URL to the Apprise Notifications box in the Basics Settings page
2. Add "Anna's Hummingbird" to the "Exclude these species (comma separated common names)" text box in the Basic Settings page
3. Play the Anna's Hummingbird call into the mic. I used the following YouTube link to test this: https://www.youtube.com/watch?v=ZhFv84PGrwA
4. Receive notification on Discord

Expected behavior
Not receiving notifications for Anna's Hummingbirds on Discord if I have it listed to be excluded

Screenshots
[image]
[image]

Additional context
This issue may not be exclusive to Discord, but it's the only platform I've tested.

Your build
Raspberry Pi 4B

Nachtzuster, 2025-02-09:
Thanks for reporting.
Looks like these settings are not being decoded properly.
Could you run `grep APPRISE_ONLY_NOTIFY_SPECIES_NAMES birdnet.conf`? if my guess is correct it should show `Anna&#39;s Hummingbird`

There's a potential fix in a testing branch, could you test the fix by doing:

```
git remote remove origin
git remote add origin https://github.com/Nachtzuster/BirdNET-Pi.git
git fetch
./scripts/update_birdnet.sh -b preserve_quotes
```
After that, rewrite the setting from the ui:
 - clear the 'Exclude these species...' box, hit 'Update Settings'
 - re-add your species list to the 'Exclude these species...' box again, hit 'Update Settings'
 

`grep APPRISE_ONLY_NOTIFY_SPECIES_NAMES birdnet.conf`  should confirm it is being written correctly now

zwk920, 2025-02-09:
I tested your fix and it seems to be working! 

I ran the ran the grep command prior to your fix and it read the apostrophe incorrectly as expected
[image]

I ran your fix and now it's reading correctly
[image]

I played some Anna's Hummingbird calls into the mic and they are being detected, but are no longer sending notifications, which is what I was looking for
[image]

Thank you! Let me know if you need any more help testing this

closed 2025-02-10 by Nachtzuster